linux-sunxi/meta-sunxi #319: u-boot patch failed: 0002-Revert-sunxi-psci-avoid-error-address-of-packed-memb.patch

TomHanlein, 2021-04-23:
Hi!

I tried to compile meta-sunxi in yocto 3.2.3 (gatesgarth) and yocto 3.3 (hardknott) for orange-pi-zero. I got the following error:
```
ERROR: u-boot-1_2021.01-r0 do_patch: Command Error: 'quilt --quiltrc /tmp/yocto/build/orange/tmp/work/orange_pi_zero-poky-linux-gnueabi/u-boot/1_2021.01-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
stdout: Applying patch 0002-Revert-sunxi-psci-avoid-error-address-of-packed-memb.patch
patching file arch/arm/cpu/armv7/sunxi/psci.c
Hunk #1 FAILED at 75.
Hunk #2 FAILED at 90.
Hunk #3 FAILED at 99.
3 out of 3 hunks FAILED -- rejects in file arch/arm/cpu/armv7/sunxi/psci.c
Patch 0002-Revert-sunxi-psci-avoid-error-address-of-packed-memb.patch can be reverse-applied

stderr:
ERROR: Logfile of failure stored in: /tmp/yocto/build/orange/tmp/work/orange_pi_zero-poky-linux-gnueabi/u-boot/1_2021.01-r0/temp/log.do_patch.12402
ERROR: Task (/tmp/yocto/poky/meta/recipes-bsp/u-boot/u-boot_2021.01.bb:do_patch) failed with exit code '1'
```
I checked the downloaded u-boot git-code in `/tmp/work/orange_pi_zero-poky-linux-gnueabi/u-boot/1_2021.01-r0/git` and it seems the patch `0002-Revert-sunxi-psci-avoid-error-address-of-packed-memb.patch` is not needed anymore.
u-boot commit `3586cb82277e8af9eae38b354bb8b2aee38ee377` has already done what this patch tries to do.

Workaround: comment the following line in the file `meta-sunxi/recipes-bsp/u-boot/u-boot_%.bbappend`
```
# SRC_URI_append_orange-pi-zero = " file://0002-Revert-sunxi-psci-avoid-error-address-of-packed-memb.patch"
```

Afterwards u-boot patches and compiles just fine.

Hope this helps anyone.

Tom

nandra, 2021-04-23:
Thanks for report. I think this change must be removed from master and keep only in dunfell. BTW are you able to compile hardknott with actual master?

nandra, 2021-04-23:
@TomHanlein fixed on master in PR #320

TomHanlein, 2021-04-23:
> BTW are you able to compile hardknott with actual master?
It is still compiling. I have to check after the weekend.

Tom

nandra, 2021-04-26:
@TomHanlein any update on hardknott build? Thanks

nandra, 2021-04-30:
Should be fixed. Closing.

closed 2021-04-30 by nandra