linux-sunxi/meta-sunxi #324: Any support for licheepi-nano?

masoudr, 2021-07-18:
Hello,
I just wondering do you have a plan to add support for [LicheePi Nano](https://www.seeedstudio.com/Sipeed-Lichee-Nano-Linux-Development-Board-16M-Flash-WiFi-Version-p-2893.html) board? It is based on F1c100s SoC.
Thank you.

nandra, 2021-08-02:
Hi, 

if you have board and can create pull request with necessary changes it would be the best. I don't own such board and adding support blindly can lead to many trials and errors. If you can support us with board we can try to add support for it in this layer. Thanks.

masoudr, 2021-08-02:
I have the board, and I've created a Buildroot config for it. The problem is that its kernel and u-boot are different from the sunxi mainline so how can I add kernel and u-boot repos to the board conf file?
Anyway, I'm quite new to Yocto, but I will try to give it a try and create a PR when I'm done.

closed 2021-12-15 by nandra

argrento, 2023-05-21:
Seems that there is no licheepi-nano support. I have the board, will try to submit a patch.

nandra, 2023-05-22:
> Seems that there is no licheepi-nano support. I have the board, will try to submit a patch.

Thanks @argrento