linux-sunxi/meta-sunxi #340: support nanopi -r1 and r2s

madmax440, 2022-03-05:
current yocto layer dont support nanopi r1 and r2s. would love to have yocto support for these boards also.

nandra, 2022-04-01:
@madmax440 if mainline kernel + u-boot have support for those boards adding it would be pretty straightforward See https://github.com/linux-sunxi/meta-sunxi/tree/master/conf/machine for inspiration pls.

madmax440, 2022-04-06:
@nandra ,  i could boot openwrt. If there is yocto layer support for nanopi r1, i am happy to test and verify !

nandra, 2022-04-07:
@madmax440 ok let me check

nandra, 2022-04-08:
@madmax440 pls try to build and boot: ```mbe-nanopi-r1``` branch. Thanks.

madmax440, 2022-04-12:
@nandra , i  have used `mbe-nanopi-r1`, but seems like it fails at the dtb build.
```bash
|   CC      arch/arm/boot/compressed/fdt_ro.o
|   CC      arch/arm/boot/compressed/fdt_wip.o
|   CC      arch/arm/boot/compressed/fdt.o
|   CC      arch/arm/boot/compressed/atags_to_fdt.o
|   SHIPPED arch/arm/boot/compressed/lib1funcs.S
|   SHIPPED arch/arm/boot/compressed/ashldi3.S
|   SHIPPED arch/arm/boot/compressed/bswapsdi2.S
|   AS      arch/arm/boot/compressed/hyp-stub.o
|   AS      arch/arm/boot/compressed/lib1funcs.o
|   AS      arch/arm/boot/compressed/ashldi3.o
|   AS      arch/arm/boot/compressed/bswapsdi2.o
|   AS      arch/arm/boot/compressed/piggy.o
|   LD      arch/arm/boot/compressed/vmlinux
|   OBJCOPY arch/arm/boot/zImage
|   Kernel: arch/arm/boot/zImage is ready
|   UIMAGE  arch/arm/boot/uImage
| Image Name:   Linux-5.10.2
| Created:      Mon Dec 21 12:30:08 2020
| Image Type:   ARM Linux Kernel Image (uncompressed)
| Data Size:    4755144 Bytes = 4643.70 KiB = 4.53 MiB
| Load Address: 40008000
| Entry Point:  40008000
|   Kernel: arch/arm/boot/uImage is ready
| NOTE: make -j 4 HOSTCC=gcc  -isystem/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/lib                         -L/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/data/nanopi/poky/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 HOSTCPP=gcc  -E HOSTCXX=g++  -isystem/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/lib                         -L/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/data/nanopi/poky/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 sun8i-h3-nanopi-r1.dtb CC=arm-poky-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fmacro-prefix-map=/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0=/usr/src/debug/linux-mainline/5.10.2-r0                      -fdebug-prefix-map=/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0=/usr/src/debug/linux-mainline/5.10.2-r0                      -fdebug-prefix-map=/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot=                      -fdebug-prefix-map=/data/nanopi/poky/build/tmp/work/nanopi_r1-poky-linux-gnueabi/linux-mainline/5.10.2-r0/recipe-sysroot-native=  -fdebug-prefix-map=/data/nanopi/poky/build/tmp/work-shared/nanopi-r1/kernel-source=/usr/src/kernel   LD=arm-poky-linux-gnueabi-ld.bfd   LOADADDR=0x40008000
|   SYNC    include/config/auto.conf.cmd
|   GEN     Makefile
| make[2]: *** No rule to make target 'arch/arm/boot/dts/sun8i-h3-nanopi-r1.dtb'.  Stop.
| make[1]: *** [/data/nanopi/poky/build/tmp/work-shared/nanopi-r1/kernel-source/Makefile:1338: sun8i-h3-nanopi-r1.dtb] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| make: *** [/data/nanopi/poky/build/tmp/work-shared/nanopi-r1/kernel-source/Makefile:185: __sub-make] Error 2
ERROR: Task (/data/nanopi/poky/meta-sunxi/recipes-kernel/linux/linux-mainline_5.10.2.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1927 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /data/nanopi/poky/meta-sunxi/recipes-kernel/linux/linux-mainline_5.10.2.bb:do_compile
Summary: There were 3 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```bash

nandra, 2022-04-13:
Pls pull form same branch and retry. I've fixed it. Thanks.

madmax440, 2022-04-13:
@nandra  Thanks I pulled and built it. Build is ok now. Good progress. Seems the boot stuck at `/dev/mmcblk0p2`. Any idea why?

```bash
U-Boot SPL 2021.07-g840658b (Jul 05 2021 - 15:11:28 +0000)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2021.07-g840658b (Jul 05 2021 - 15:11:28 +0000) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyARM NanoPi R1
DRAM:  512 MiB
MMC:   mmc@1c0f000: 0, mmc@1c11000: 1
Loading Environment from FAT... Card did not respond to voltage select! : -110
In:    serial
Out:   serial
Err:   serial
Net:   phy interface7

Error: ethernet@1c30000 address not set.
No ethernet found.

starting USB...
Bus usb@1c1a000: USB EHCI 1.00
Bus usb@1c1a400: USB OHCI 1.0
Bus usb@1c1d000: USB EHCI 1.00
Bus usb@1c1d400: USB OHCI 1.0
scanning bus usb@1c1a000 for devices... 1 USB Device(s) found
scanning bus usb@1c1a400 for devices... 1 USB Device(s) found
scanning bus usb@1c1d000 for devices... 1 USB Device(s) found
scanning bus usb@1c1d400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
714 bytes read in 2 ms (348.6 KiB/s)
## Executing script at 43100000
23711 bytes read in 4 ms (5.7 MiB/s)
Failed to load 'zImage'
Failed to load 'boot/zImage'
4755208 bytes read in 338 ms (13.4 MiB/s)
zimage: Bad magic!
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-5.10.2
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4755144 Bytes = 4.5 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
EHCI failed to shut down host controller.
   Loading Kernel Image
   Loading Device Tree to 49ff7000, end 49fffc9e ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.2 (oe-user@oe-host) (arm-poky-linux-gnueabi-gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.37.20210721) #1 SMP Mon Dec 21 12:30:08 UTC 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: FriendlyARM NanoPi R1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[... 181 lines cut ...]
[    1.614796] ehci-platform 1c1d000.usb: USB 2.0 started, EHCI 1.00
[    1.622079] hub 3-0:1.0: USB hub found
[    1.625999] hub 3-0:1.0: 1 port detected
[    1.631417] ohci-platform 1c1b400.usb: Generic Platform OHCI controller
[    1.638154] ohci-platform 1c1b400.usb: new USB bus registered, assigned bus number 4
[    1.646087] ohci-platform 1c1b400.usb: irq 38, io mem 0x01c1b400
[    1.719834] hub 4-0:1.0: USB hub found
[    1.723682] hub 4-0:1.0: 1 port detected
[    1.729586] ohci-platform 1c1c400.usb: Generic Platform OHCI controller
[    1.736355] ohci-platform 1c1c400.usb: new USB bus registered, assigned bus number 5
[    1.744263] ohci-platform 1c1c400.usb: irq 40, io mem 0x01c1c400
[    1.819832] hub 5-0:1.0: USB hub found
[    1.823687] hub 5-0:1.0: 1 port detected
[    1.829234] ohci-platform 1c1d400.usb: Generic Platform OHCI controller
[    1.835978] ohci-platform 1c1d400.usb: new USB bus registered, assigned bus number 6
[    1.843880] ohci-platform 1c1d400.usb: irq 42, io mem 0x01c1d400
[    1.874645] usb 1-1: new high-speed USB device number 2 using ehci-platform
[    1.919875] hub 6-0:1.0: USB hub found
[    1.923732] hub 6-0:1.0: 1 port detected
[    1.944383] cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequency: 480000 KHz, changing to: 648000 KHz
[    1.956523] Waiting for root device /dev/mmcblk0p2...

CTRL-A Z for help | 115200 8N1 | NOR | Minicom 2.8 | VT102 | Online 0:1 | ttyUSB0                                                                                      

````

madmax440, 2022-04-13:
@nandra after i use a new SD card, the system boots. 

There is still another problem. The nanopi-r1 contains two ethernets. However all i see is one ethernet in `ifconfig`.

Can we fix that too?

```bash
root@nanopi-r1:~# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 86:94:B6:6D:A8:BE  
          inet addr:192.168.1.100  Bcast:0.0.0.0  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:46 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

nandra, 2022-04-14:
@madmax440 unfortunately in mainline kernel is support for only on interface so I cannot help much here. Does openwrt support both? Thanks.

madmax440, 2022-04-15:
@nandra Yes openwrt has dual ethernet support. Please see below from the console.

```bash
root@OpenWrt:~# cat /etc/banner
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r17648+1-16e83a7491
 -----------------------------------------------------
root@OpenWrt:~# ifconfig 
eth0      Link encap:Ethernet  HWaddr 96:05:F4:4A:7B:0B  
          inet addr:192.168.1.100  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::9405:f4ff:fe4a:7b0b/64 Scope:Link
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1500  Metric:1
          RX packets:480 errors:0 dropped:0 overruns:0 frame:0
          TX packets:444 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:449224 (438.6 KiB)  TX bytes:83538 (81.5 KiB)
          Interrupt:36 

eth1      Link encap:Ethernet  HWaddr A6:E9:20:18:EB:52  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:634 (634.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:424 errors:0 dropped:0 overruns:0 frame:0
          TX packets:424 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:36364 (35.5 KiB)  TX bytes:36364 (35.5 KiB)

root@OpenWrt:~# 
```

Thanks for your support and time so far. Wish there is a yocto port.

nandra, 2022-04-21:
OK I don't have capacity atm to work on it so please feel free to post patch. I'm closing the issue. Thanks.

closed 2022-04-21 by nandra