linux-sunxi/meta-sunxi #342: How to utilize device tree overlays

MitchGaines, 2022-04-01:
Hi there, 

I'm finding that when I use this layer, I'm unable to utilize the GPIOs on my board. It would seem I need to utilize dt overlays to make them work, but I can't seem to find any documentation on how to properly do this. It doesn't seem like this is supported. How would one go about creating dtoverlays to permit the use of the GPIO devices? 

Specifically, I'm using an orange pi +2E and am setting the appropriate dtb by setting `KERNEL_DEVICETREE="sun8i-h3-orangepi-plus2e.dtb"`. All of this works. The kernel boots up without issue, but I try to communicate with the GPIOs and it doesn't work. Even hooked up an oscope to the pins trying to figure out what was going on and no signals were being sent. I assume I'm doing something wrong here, so would appreciate being pointed in the right direction. 

Thanks for the help!

nandra, 2022-04-03:
Hi,

which GPIO you would like to use? GPIO's exported on header can be used from userspace. Can you please be more specific. Thanks.

MitchGaines, 2022-04-05:
Hi yes, thank you for the response. Below I will add a number of details and what I have done in terms of troubleshooting.

So, I have 2 interfaces I am attempting to control: i2c-0 and uart2. Both of them appear to have the same issue. Though, all of my troubleshooting has focused on i2c-0. I have a device driver that is attempting to write to the i2c device, but is failing with the following: `ENXIO (No such device or address)`. This device driver has no problem executing on Armbian-based platforms running on the same device with the same parameters. Further, it runs without issue on a Raspberry Pi when I cross compile and swap the i2c interface. Point is, the device driver works on everything except for my yocto-based Orange Pi +2E. When I start working with the `i2c-tools` to troubleshoot (I'll post some data from this below), I run into the same issue. So, I think this is a system issue, not a driver issue. 

On the topic of i2c-tools. One an Armbian device that works, I run the following with the corresponding return.
```
% sudo i2cdetect 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0.
I will probe address range 0x03-0x77.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- 44 -- -- -- -- -- -- 4b -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```

Conversely, when I run on the yocto-based system, I get the following:
```
# i2cdetect 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0.
I will probe address range 0x03-0x77.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- 65 -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --      
```

Any attempt to read or write from the i2c device results in `ENXIO (No such device or address)`. That specific error would indicate that the system believes it is sending the message, but not receiving an ACK in response. Out of curiosity, I hooked up an oscilloscope to see if any messages were being sent on the pins. No data was being sent. 

The reason I mention device tree overlays is because I know on a lot of embedded linux systems, you'll have a file which defines which devices you want enabled. That doesn't seem to be the case here, but you say all the GPIOs _should_ work in user space. Are there kernel configs I need to be adding? I tried adding `CONFIG_I2C=y` and `CONFIG_I2C_CHARDEV=y` to my kernel config to no avail. I'm kind of at a loss as to why the system wouldn't even be sending signals to the GPIOs at this point. 

Really appreciate the help here. Please let me know if you have any questions or suggestions.

nandra, 2022-04-06:
@MitchGaines I would say difference is in kernel version probably. Can you please share which kernel you use on armbian and which from yocto. Also I would check in yocto also ```i2cdetect 1```. Sometimes numbering is bit tricky. Please also share yocto bootlog (output of ```dmesg``` command). Thanks.

MitchGaines, 2022-04-06:
@nandra 

So, running `i2cdetect 1` returns `Error: Could not open file '/dev/i2c-1' or '/dev/i2c/1': No such file or directory`, which I presume makes sense. But, I do understand wanting to test that. 

The Armbian system which works is running kernel version `4.19.62-sunxi` whereas my yocto system is on the `dunfell` branch and running `5.4.111` of linux-mainline.

Here is the full `dmesg` output:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.111 (oe-user@oe-host) (gcc version 9.3.0 (GCC)) #1 SMP Sat Apr 10 11:34:32 UTC 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Xunlong Orange Pi Plus 2E
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 524288
[    0.000000]   Normal zone: 1536 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 327680 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 15 pages/cpu s31436 r8192 d21812 u61440
[    0.000000] pcpu-alloc: s31436 r8192 d21812 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 522752
[    0.000000] Kernel command line: console=ttyS0,115200 console=tty1 root=/dev/ rootwait panic=10 root=/dev/mmcblk0p3 rauc.slot=B
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 2068716K/2097152K available (7168K kernel code, 419K rwdata, 1512K rodata, 1024K init, 250K bss, 28436K reserved, 0K cma-reserved, 1310708K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2b0/0x460 with crng_init=0
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000016] Switching to timer-based delay loop, resolution 41ns
[    0.000181] Console: colour dummy device 80x30
[    0.000476] printk: console [tty1] enabled
[    0.000523] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000551] pid_max: default: 32768 minimum: 301
[    0.000684] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000711] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001437] CPU: Testing write buffer coherency: ok
[    0.001799] /cpus/cpu@0 missing clock-frequency property
[    0.001828] /cpus/cpu@1 missing clock-frequency property
[    0.001850] /cpus/cpu@2 missing clock-frequency property
[    0.001873] /cpus/cpu@3 missing clock-frequency property
[    0.001890] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002389] Setting up static identity map for 0x40100000 - 0x40100060
[    0.002521] rcu: Hierarchical SRCU implementation.
[    0.002984] smp: Bringing up secondary CPUs ...
[    0.013684] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.024465] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.035178] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.035264] smp: Brought up 1 node, 4 CPUs
[    0.035319] SMP: Total of 4 processors activated (192.00 BogoMIPS).
[    0.035333] CPU: All CPU(s) started in HYP mode.
[    0.035344] CPU: Virtualization extensions available.
[    0.036064] devtmpfs: initialized
[    0.040449] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.040682] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.040717] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[... 221 lines cut ...]
[    3.795288] systemd[1]: Dependency failed for My Monitor.
[    3.801734] systemd[1]: my-monitor.service: Job my-monitor.service/start failed with result 'dependency'.
[    3.814827] systemd[1]: Started Journal Service.
[    3.872663] EXT4-fs (mmcblk0p3): re-mounted. Opts: (null)
[    3.910784] systemd-journald[111]: Received client request to flush runtime journal.
[    5.307439] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    8.340324] random: crng init done
[    8.343760] random: 7 urandom warning(s) missed due to ratelimiting
[... cut ...]
[    9.791245] EXT4-fs (mmcblk0p2): ext4_check_descriptors: Checksum for group 0 failed (54896!=15185)
[    9.791257] EXT4-fs (mmcblk0p2): group descriptors corrupted!
[   14.480887] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   14.480933] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   15.348982] dwmac-sun8i 1c30000.ethernet eth0: Too many address, switching to promiscuous
[   32.480328] vcc3v0: disabling
[   32.480346] vcc5v0: disabling
[   32.480352] usb0-vbus: disabling

```

Also, really appreciate the help thus far. Thank you.

nandra, 2022-04-08:
Please try this patch and see if it will work:
```diff --git a/arch/arm/boot/dts/sun8i-h3-orangepi-plus2e.dts b/arch/arm/boot/dts/sun8i-h3-orangepi-plus2e.dts
index b6ca45d..c0a3b11 100644
--- a/arch/arm/boot/dts/sun8i-h3-orangepi-plus2e.dts
+++ b/arch/arm/boot/dts/sun8i-h3-orangepi-plus2e.dts
@@ -77,3 +77,12 @@
                reg = <1>;
        };
 };
+
+&i2c0 {
+       status = "okay";
+};
+
+&i2c1 {
+       status = "okay";
+};
+
```
Thanks.

MitchGaines, 2022-04-08:
AHA. That did it. Thanks so much for the help, @nandra. Really appreciate it.

closed 2022-04-08 by MitchGaines