linux-sunxi/meta-sunxi #343: Missing layer dependency in documentation?

xtrinch, 2022-06-22:
It appears that in the kirkstone branch, the `/meta-openembedded/meta-python` needs to be added as a layer dependency, otherwise it throws parsing errors (`ERROR: ParseError at **/meta-sunxi/recipes-devtools/python/pya20_0.2.12.bb:16: Could not inherit file classes/distutils3.bbclass`).

In the dunfell branch, this didn't need to be added.

It is also throwing deprecation errors on kirkstone: ```WARNING: /home/xtrinch/repos/yocto-a10-olinuxino-lime/meta-sunxi/recipes-devtools/python/pya20_0.2.12.bb: distutils-common-base.bbclass is deprecated, please use setuptools3-base.bbclass instead```

nandra, 2022-06-27:
I checked by building `py20` recipe but I don't see any error just warning about `distutils3` deprecation. Can you please share your `bblayers.conf` ? Thanks.

xtrinch, 2022-06-27:
As a quick side question, the base sato image builds fine, but when I tried to add the Epiphany browser into the base sato image, it's throwing the following error:

```
DISTRO_FEATURES += " x11 opengl"
IMAGE_INSTALL:append = " epiphany"
```

```
| Checking if "-mtls-dialect=gnu2" runs: YES
| 
| ../mesa-22.0.3/meson.build:555:6: ERROR: Problem encountered: xlib based GLX requires at least one gallium driver
| 
```

Is this in any way related to the meta-sunxi?

nandra, 2022-06-27:
Did you try to enable lima?

BTW I cannot reproduce issue you have. It print warning but if I switch to setuptools pya20 doesn't compile at all.

nandra, 2022-09-09:
Fixed with #344

closed 2022-09-09 by nandra