linux-sunxi/meta-sunxi #356: core-image-base build fail for olinuxino-a13 machine on Kirkstone branch

spanceac, 2022-12-15:
Hi,

I am unable to build `core-image-base` for `olinuxino-a13` using `kirkstone` branch.
Build command:

`MACHINE=olinuxino-a13 bitbake core-image-base`

Build error log:
`ERROR: mesa-gl-2_22.0.3-r0 do_configure: meson failed
ERROR: mesa-gl-2_22.0.3-r0 do_configure: ExecutionError('/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/temp/run.do_configure.339046', 1, None, None)
ERROR: Logfile of failure stored in: /home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/temp/log.do_configure.339046
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-core/expat/expat_2.5.0.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-core/zlib/zlib_1.2.11.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/gcc/gcc-cross_11.3.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/gcc/gcc-runtime_11.3.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/qemu/qemu-native_6.2.0.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-app/xrandr_1.5.1.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-lib/libx11_1.7.3.1.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-lib/libxdamage_1.1.5.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-lib/libxext_1.3.4.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-lib/libxfixes_6.0.0.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-lib/libxxf86vm_1.1.4.bb:do_populate_sysroot', '/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-proto/xorgproto_2021.5.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-core/gettext/gettext_0.21.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-core/libxml/libxml2_2.9.14.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/bison/bison_3.8.2.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/chrpath/chrpath_0.16.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/flex/flex_2.6.4.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/meson/meson_0.61.3.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/pkgconfig/pkgconfig_git.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/python/python3-mako_1.1.6.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-devtools/python/python3_3.10.7.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-extended/xz/xz_5.2.6.bb:do_populate_sysroot', 'virtual:native:/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/xorg-util/makedepend_1.0.6.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['expat', 'glibc', 'zlib', 'gcc-cross-arm', 'gcc-runtime', 'qemu-native', 'quilt-native', 'xrandr', 'libx11', 'libxdamage', 'libxext', 'libxfixes', 'libxxf86vm', 'xorgproto', 'gettext-native', 'libxml2-native', 'bison-native', 'chrpath-native', 'flex-native', 'meson-native', 'ninja-native', 'patch-native', 'pkgconfig-native', 'pseudo-native', 'python3-mako-native', 'python3-native', 'xz-native', 'makedepend-native', 'python3-wheel-native', 'python3-setuptools-native', 'python3-installer-native', 'libtool-native', 'attr-native', 'util-macros', 'texinfo-dummy-native', 'gettext-minimal-native', 'zlib-native', 'm4-native', 'readline-native', 'gdbm-native', 'bzip2-native', 'sqlite3-native', 'util-linux-libuuid-native', 'libffi-native', 'libnsl2-native', 'libtirpc-native', 'ncurses-native', 'openssl-native', 're2c-native', 'libgcc', 'linux-libc-headers', 'glib-2.0-native', 'util-macros-native', 'xorgproto-native', 'libxrender', 'libxrandr', 'gnu-config-native', 'binutils-cross-arm', 'gmp-native', 'libmpc-native', 'zstd-native', 'mpfr-native', 'libxcb', 'autoconf-archive', 'xtrans', 'python3-flit-core-native', 'unzip-native', 'perl-native', 'libpcre-native', 'util-linux-native', 'xcb-proto', 'libxau', 'libxdmcp', 'libpthread-stubs', 'make-native', 'libcap-ng-native', 'libpcre2-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| NOTE: Executing meson -Dshared-glapi=enabled -Dglx-read-only-text=true -Dplatforms=x11 -Ddri3=disabled -Degl=disabled -Delf-tls=false -Dgallium-drivers= -Dllvm=disabled -Dgbm=disabled -Dgles1=disabled -Dgles2=disabled -Dlmsensors=disabled -Dgallium-opencl=disabled -Dopencl-spirv=false -Dopengl=true -Dosmesa=false -Dlibunwind=disabled -Dgallium-va=disabled -Dgallium-vdpau=disabled -Dvulkan-drivers= -Dgallium-xa=disabled -Dgallium-xvmc=disabled...
| The Meson build system
| Version: 0.61.3
| Source dir: /home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/mesa-22.0.3
| Build dir: /home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/build
| Build type: cross build
| Program python3 found: YES (/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/recipe-sysroot-native/usr/bin/python3-native/python3)
| Project name: mesa
| Project version: 22.0.3
| C compiler for the host machine: arm-poky-linux-gnueabi-gcc -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a8 -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/recipe-sysroot (gcc 11.3.0 "arm-poky-linux-gnueabi-gcc (GCC) 11.3.0")
| C linker for the host machine: arm-poky-linux-gnueabi-gcc -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a8 -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/recipe-sysroot ld.bfd 2.38.20220708
| C++ compiler for the host machine: arm-poky-linux-gnueabi-g++ -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a8 -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/recipe-sysroot (gcc 11.3.0 "arm-poky-linux-gnueabi-g++ (GCC) 11.3.0")
| C++ linker for the host machine: arm-poky-linux-gnueabi-g++ -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a8 -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/mesa-gl/2_22.0.3-r0/rec [... cut ...]
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/mesa/mesa-gl_22.0.3.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2038 tasks of which 2022 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/spanceac/sebu/pocketbook/yocto-pocketbook/poky/meta/recipes-graphics/mesa/mesa-gl_22.0.3.bb:do_configure
Summary: There were 3 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
`

nandra, 2022-12-15:
@spanceac let me try. Did you maybe tried master if it works there please? Thanks.

spanceac, 2022-12-17:
I will try it and let you know

spanceac, 2022-12-17:
@nandra I tried with the following combo:
* poky on kirkstone branch
* meta-openembedded on kirkstone branch
* meta-sunxi on the master branch

The same error shows up.

nandra, 2022-12-22:
OK thanks for testing. Will try to build today and check. Thanks.

nandra, 2022-12-22:
@spanceac works fine here on kirkstone also on master. Did you try to wipe out build/tmp directory pls? Thanks.

spanceac, 2022-12-23:
I will try with everything clean and will let you know. Maybe it's an env issue on my side

spanceac, 2022-12-23:
@nandra Even with a clean clone of layers and clean builds the issue still reproduces.

This is the clone and build procedure that I use:
```
git clone git://git.yoctoproject.org/poky
cd poky
git checkout kirkstone
cd ..
git clone https://github.com/linux-sunxi/meta-sunxi
cd meta-sunxi
git checkout kirkstone
cd ..
git clone https://github.com/openembedded/meta-openembedded
cd meta-openembedded
git checkout kirkstone
cd ..
source oe-init-build-env
bitbake-layers add-layer ../meta-openembedded/meta-oe
bitbake-layers add-layer ../meta-openembedded/meta-python/
bitbake-layers add-layer ../meta-sunxi/
MACHINE=olinuxino-a13 bitbake core-image-base
```

nandra, 2023-01-10:
@spanceac did you found anything in this regard which resolve an issue? Which OS do you use. You tried to build in docker container maybe? I'm using Ubuntu 20.04

spanceac, 2023-01-12:
hi @nandra! 

I'm using `Linux Mint 21 Vanessa`  which is based on Ubuntu 22.04. I will also try with a 20.04 VM and let you know

spanceac, 2023-01-12:
@nandra the exact same issue happens in an Ubuntu 20.04 VM

```
sebi@sebi-Standard-PC-i440FX-PIIX-1996:~/build/poky/build$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.5 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

ludiazv, 2023-01-15:
Same error on a bpi m2 zero.  Solved adding ```MACHINEOVERRIDES .= ":use-mailine-graphics"``` in local.conf

nandra, 2023-01-16:
@ludiazv : good catch :+1: . I have it also in my default config. @spanceac pls try to above. Thanks.

spanceac, 2023-01-18:
I confirm that adding `MACHINEOVERRIDES .= ":use-mailine-graphics"` in local.conf of build prevents the error from appearing. IMO, this either has to be documented or a proper fix is needed. I guess there's a reason why `sunxi-mali` stuff is trying to be used instead of mainline graphics

nandra, 2023-01-18:
#359 kind of related. I'll try to come up with some fix. Thanks for testing @spanceac

nandra, 2023-01-23:
@spanceac please see #360 You can try to drop `MACHINEOVERRIDES .= ":use-mailine-graphics"` from local.conf

nandra, 2023-01-24:
@spanceac can we please close this ticket? Thanks.

nandra, 2023-01-25:
Closing. Changes merged.

closed 2023-01-25 by nandra

spanceac, 2023-01-25:
Thank you for the support @nandra. I tested it just now and it worked