linux-sunxi/meta-sunxi #359: core-image-sato kirkstone bananapi

voloviq, 2023-01-14:
Hi,
again my problem is back with the compilation of core-image-sato.
Now the problem is related to package
xf86-video-fbturbo_git.bb which needs sunxi-mali
I try to build only sunxi-mali but with the following error

`ERROR: Nothing PROVIDES 'sunxi-mali'

sunxi-mali was skipped: PREFERRED_PROVIDER_virtual/egl set to mesa, not sunxi-mali
`
How can I handle this issue?

nandra, 2023-01-16:
Hi @voloviq , can you plaase specify which branch you use for testing?

Also pls try to add `MACHINEOVERRIDES .= ":use-mailine-graphics"` to your local.conf

voloviq, 2023-01-16:
Hi Marek,

I use kirkstone, of course, I added MACHINEOVERRIDES .= ":use-mailine-graphics" in my local.

BR Michal

nandra, 2023-01-16:
Hi Michał, let me try to build it. I'll get back when have something.

nandra, 2023-01-17:
@voloviq : I'm able to reproduce. Looking into it. Let you know.

voloviq, 2023-01-17:
@nandra Thanks I will wait, I realize that this is not a simple issue.

voloviq, 2023-01-23:
Hi Marek,

Compilation passed, unfortunately xorg not started

> X.Org X Server 1.21.1.4
> X Protocol Version 11, Revision 0
> Current Operating System: Linux bananapi 5.15.35 #1 SMP Wed Apr 20 07:34:22 UTC 2022 armv7l
> Kernel command line: console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootwait panic=10
>  
> Current version of pixman: 0.40.0
>         Before reporting problems, check http://wiki.x.org
>         to make sure that you have the latest version.
> Markers: (--) probed, (**) from config file, (==) default setting,
>         (++) from command line, (!!) notice, (II) informational,
>         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
> (==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  9 12:34:58 2018
> (==) Using system config directory "/usr/share/X11/xorg.conf.d"
> (EE) 
> Fatal server error:
> (EE) no screens found(EE) 
> (EE) 
> Please consult the The X.Org Foundation support 
>          at http://wiki.x.org
>  for help. 
> (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
> (EE) 
> (EE) Server terminated with error (1). Closing log file.

Content of Xorg.0.log

> [     8.016] 
> X.Org X Server 1.21.1.4
> X Protocol Version 11, Revision 0
> [     8.016] Current Operating System: Linux bananapi 5.15.35 #1 SMP Wed Apr 20 07:34:22 UTC 2022 armv7l
> [     8.016] Kernel command line: console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootwait panic=10
> [     8.016]  
> [     8.016] Current version of pixman: 0.40.0
> [     8.016]    Before reporting problems, check http://wiki.x.org
>         to make sure that you have the latest version.
> [     8.016] Markers: (--) probed, (**) from config file, (==) default setting,
>         (++) from command line, (!!) notice, (II) informational,
>         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
> [     8.016] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  9 12:34:58 2018
> [     8.027] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
> [     8.031] (==) No Layout section.  Using the first Screen section.
> [     8.031] (==) No screen section available. Using defaults.
> [     8.031] (**) |-->Screen "Default Screen Section" (0)
> [     8.031] (**) |   |-->Monitor "<default monitor>"
> [     8.033] (==) No monitor specified for screen "Default Screen Section".
>         Using a default monitor configuration.
> [     8.034] (==) Automatically adding devices
> [     8.034] (==) Automatically enabling devices
> [     8.034] (==) Automatically adding GPU devices
> [     8.034] (==) Automatically binding GPU devices
> [     8.038] (==) Max clients allowed: 256, resource mask: 0x1fffff
> [     8.039] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
> [     8.039]    Entry deleted from font path.
> [     8.040] (WW) The directory "/usr/share/fonts/X11/TTF" does not exist.
> [     8.040]    Entry deleted from font path.
> [     8.040] (WW) The directory "/usr/share/fonts/X11/OTF" does not exist.
> [     8.040]    Entry deleted from font path.
> [     8.041] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
> [     8.041]    Entry deleted from font path.
> [     8.041] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
> [     8.041]    Entry deleted from font path.
> [     8.041] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
> [     8.041]    Entry deleted from font path.
> [     8.041] (==) FontPath set to:
> 
> [     8.041] (==) ModulePath set to "/usr/lib/xorg/modules"
> [     8.041] (II) The server relies on udev to provide the list of input devices.
>         If no devices become available, reconfigure udev or disable AutoAddDevices.
> [     8.041] (II) Module ABI versions:
> [     8.041]    X.Org ANSI C Emulation: 0.4
> [     8.041]    X.Org Video Driver: 25.2
> [     8.041]    X.Org XInput driver : 24.4
> [     8.041]    X.Org Server Extension : 10.0
> [     8.043] (II) no primary bus or device found
> [     8.043] (II) LoadModule: "glx"
> [     8.047] (WW) Warning, couldn't open module glx
> [     8.047] (EE) Failed to load module "glx" (module does not exist, 0)
> [     8.048] (==) Matched modesetting as autoconfigured driver 0
> [     8.048] (==) Matched fbdev as autoconfigured driver 1
> [     8.048] (==) Assigned the driver to the xf86ConfigLayout
> [     8.048] (II) LoadModule: "modesetting"
> [     8.048] (WW) Warning, couldn't open module modesetting
> [     8.048] (EE) Failed to load module "modesetting" (module does not exist, 0)
> [     8.048] (II) LoadModule: "fbdev"
> [     8.049] (WW) Warning, couldn't open module fbdev
> [     8.049] (EE) Failed to load module "fbdev" (module does not exist, 0)
> [     8.049] (EE) No drivers available.
> [     8.049] (EE) 
> Fatal server error:
> [     8.049] (EE) no screens found(EE) 
> [     8.050] (EE) 
> Please consult the The X.Org Foundation support 
>          at http://wiki.x.org
>  for help. 
> [     8.051] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
> [     8.051] (EE) 
> [     8.051] (EE) Server terminated with error (1). Closing log file.
>

voloviq, 2023-01-24:
Yes You right, I add this information that on meta-sunxi kernel defconfig driver is missing. I think We can close this issue. Thanks a lot.

closed 2023-01-24 by voloviq

nandra, 2023-01-24:
@voloviq ok I'll merge #360 to master and also to kirkstone branch.