linux-sunxi/meta-sunxi #367: Communication platform

nandra, 2023-02-08:
Hi, this is just an idea and brainstorming to speed up communication. I'm thinking about something like IRC channel for sunxi related things or maybe slack/gitter? Any other ideas? Please comment. Thanks.

pblxptr, 2023-02-22:
@nandra  sounds good to me, either IRC, slack, gitter, or even discord.

slapin, 2023-02-22:
I guess we have some communication channels already
https://linux-sunxi.org/IRC.
But I'm fine with new IRC too.

nandra, 2023-02-22:
@slapin was not aware about that that IRC already exists but it's more general then related to yocto only or? Thanks.

pblxptr, 2023-02-23:
Have you decided what to use? Btw. @nandra you asked to add also support for OPIZ2 for mainline Kernel 6x. even if some peripherals are not working - I did this, but I haven't tested it yet. Will do this after work, probably today evening. Moreover, I managed to compile wifi driver for Kernel 6.1.2, ofc it doesn't mean it will work but at least it compiles, so I will try to run it, but I can't promise anything because this driver is really huge in terms of lines of code and complexity. Anyway, we will see, maybe we will be lucky.

pblxptr, 2023-03-08:
@nandra  to which target branch should I submit my changes that introduce support for OPIZ2 for 6x kernel? I tested it with kirkstone and it seems to be working fine whereas in the case of  e.g. langdale release I'm getting an error while building ATF, probably due to the change of binutils version in langdale release (2.38 -> 2.39) where they introduced a few more warnings. Probably related to https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/19401. 
I can also patch ATF to skip this check. What do you think?

nandra, 2023-03-09:
@pblxptr I think on top of master pls and add patch for atf so we can skip some warnings. Thanks.

pblxptr, 2023-03-10:
> @pblxptr I think on top of master pls and add patch for atf so we can skip some warnings. Thanks.

Oki, I should prepare it soon (hopefully next week) as I've already have it almost working - by almost I mean I've also ported a wifi driver and I'm testing it. Looks promising as I've managed to establish connection etc.

nandra, 2023-04-04:
IN the end I've created space on gitter. https://matrix.to/#/#meta-sunxi:gitter.im Feel free to join. I'll update also Readme with this info.

closed 2023-04-04 by nandra