linux-sunxi/meta-sunxi #368: ATF Build failed for OrangePi PC2

javad123javad, 2023-03-13:
Hi,
When trying to build `core-image-minimal` for OrangePi PC2, the `atf` build failed with the following error:
```
make: Entering directory '/media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/git'

  LD      /media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/build/sun50iw1p1/release/bl31/bl31.elf

Building sun50iw1p1

aarch64-poky-linux-ld: warning: /media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/build/sun50iw1p1/release/bl31/bl31.elf has a LOAD segment with RWX permissions

make: *** [Makefile:448: /media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/build/sun50iw1p1/release/bl31/bl31.elf] Error 1

make: Leaving directory '/media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/git'
```
Full backlog:
```
Loading cache: 100% |##########################################################################################################################################################################################################| Time: 0:00:00
Loaded 3690 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################################################################| Time: 0:00:00
Parsing of 2279 .bb files complete (2278 cached, 1 parsed). 3690 targets, 118 skipped, 0 masked, 0 errors.
NOTE: selecting u-boot to satisfy virtual/bootloader due to PREFERRED_PROVIDERS
NOTE: Resolving any missing task queue dependencies
NOTE: selecting binutils-cross-aarch64 to satisfy virtual/aarch64-poky-linux-binutils due to PREFERRED_PROVIDERS
NOTE: selecting pseudo-native to satisfy virtual/fakeroot-native due to PREFERRED_PROVIDERS
NOTE: selecting glibc to satisfy virtual/libc due to PREFERRED_PROVIDERS
NOTE: selecting openssl-native to satisfy openssl-native due to PREFERRED_PROVIDERS
NOTE: selecting gcc-cross-aarch64 to satisfy virtual/aarch64-poky-linux-gcc due to PREFERRED_PROVIDERS
NOTE: selecting pkgconfig-native to satisfy pkgconfig-native due to PREFERRED_PROVIDERS
NOTE: selecting gcc-runtime to satisfy virtual/aarch64-poky-linux-compilerlibs due to PREFERRED_PROVIDERS
NOTE: selecting u-boot to satisfy runtime u-boot due to PREFERRED_PROVIDER_u-boot = u-boot
NOTE: selecting u-boot to satisfy runtime u-boot-env due to PREFERRED_PROVIDER_u-boot = u-boot
NOTE: selecting u-boot to satisfy runtime u-boot-dev due to PREFERRED_PROVIDER_u-boot = u-boot
NOTE: selecting glibc to satisfy runtime glibc-utils due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting glibc to satisfy runtime ldconfig due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting glibc to satisfy runtime glibc-dev due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting linux-libc-headers to satisfy runtime linux-libc-headers-dev due to PREFERRED_PROVIDER_linux-libc-headers = linux-libc-headers
NOTE: selecting glibc to satisfy runtime glibc due to PREFERRED_PROVIDER_virtual/libc = glibc
NOTE: selecting linux-libc-headers to satisfy linux-libc-headers due to PREFERRED_PROVIDERS
NOTE: selecting libgcc to satisfy libgcc due to PREFERRED_PROVIDERS
NOTE: selecting gcc-cross-aarch64 to satisfy virtual/aarch64-poky-linux-g++ due to PREFERRED_PROVIDERS
NOTE: selecting libgcc to satisfy runtime libgcc due to PREFERRED_PROVIDER_libgcc = libgcc
NOTE: selecting glibc-locale to satisfy runtime glibc-gconvs due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting glibc-locale to satisfy runtime glibc-charmaps due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale
NOTE: selecting glibc-locale to satisfy runtime glibc-localedatas due to PREFERRED_PROVIDER_virtual/libc-locale = glibc-locale

Build Configuration:
BB_VERSION           = "2.2.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "orange-pi-pc2"
DISTRO               = "poky"
DISTRO_VERSION       = "4.1.3"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "langdale:91d0157d6daf4ea61d6b4e090c0b682d3f3ca60f"
meta-oe              
meta-python          = "langdale:b5b732876da1885ecbab2aa45f80d7a3086c5262"
meta-sunxi           = "master:1a42a71bb1a1898679d4f8e98166eeaa1db59f67"

NOTE: Marking Active Tasks###################################################################################                                                                                                                  | ETA:  0:00:00
NOTE: Pruned 1512 inactive tasks, 650 left####################################################################                                                                                                                 | ETA:  0:00:00
NOTE: Assign Weightings
NOTE: Compute totals (have 1 endpoint(s))#####################################################################                                                                                                                 | ETA:  0:00:00
Initialising tasks: 100% |#####################################################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 1 Local 0 Mirrors 0 Missed 1 Current 145 (0% match, 99% complete)
NOTE: Executing Tasks
[... 24 lines cut ...]
ERROR: atf-sunxi-git-r0 do_compile: oe_runmake failed
WARNING: exit code 1 from a shell command.

+ die oe_runmake failed
+ bbfatal_log oe_runmake failed
+ [ -p /media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/temp/fifo.1926419 ]
+ printf %b\0 bbfatal_log oe_runmake failed
+ exit 1
+ bb_sh_exit_handler
+ ret=1
+ [ 1 != 0 ]
+ echo WARNING: exit code 1 from a shell command.
+ exit 1

ERROR: atf-sunxi-git-r0 do_compile: ExecutionError('/media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/temp/run.do_compile.1926419', 1, None, None)
ERROR: Logfile of failure stored in: /media/Embedded/yocto/poky/build/tmp/work/cortexa53-crypto-poky-linux/atf-sunxi/git-r0/temp/log.do_compile.1926419
ERROR: Task (/media/Embedded/yocto/poky/meta-sunxi/recipes-bsp/atf/atf-sunxi_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 645 tasks of which 644 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /media/Embedded/yocto/poky/meta-sunxi/recipes-bsp/atf/atf-sunxi_git.bb:do_compile
Summary: There was 1 WARNING message.
Summary: There were 2 ERROR messages, returning a non-zero exit code.

```

pblxptr, 2023-03-13:
That is the problem I mentioned in the other thread. Related to: https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/19401
As far as I know, In the langdale release, they have changed the binutils version from 2.38 -> 2.39, where 2.39 contains more checks. 

I have already prepared a patch for this, but I was going to include it in PR with support for OPIZ2 for Kernel 6x (still in progress, I've still been testing some things), but I can deliver it separately.

javad123javad, 2023-03-14:
> That is the problem I mentioned in the other thread. Related to: https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/19401 As far as I know, In the langdale release, they have changed the binutils version from 2.38 -> 2.39, where 2.39 contains more checks.
> 
> I have already prepared a patch for this, but I was going to include it in PR with support for OPIZ2 for Kernel 6x (still in progress, I'm still testing some things), but I can deliver it separately.
> 
> Edit: They merged it to the 'integration' branch today. I'm not familiar with the development process of ATF but for me - if it was merged into the 'integration' branch it means that sooner or later it will be available in the mainline. The problem is that we still don't use the mainline ATF - except in the case of OPIZ2 available on 'kirkstone' branch and soon on master, so a patch is needed anyway unless we completely switch to mainline ATF.

Thanks,
I've rolled back to `dunfell` and manually added support for OPC2, until a stable release become available in mainline.

pblxptr, 2023-03-14:
> Thanks, I've rolled back to `dunfell` and manually added support for OPC2, until a stable release become available in mainline.

You have OPC2 available on `kirkstone` branch and on this release it should work fine. Give it a try :)

javad123javad, 2023-03-14:
> > > That is the problem I mentioned in the other thread. Related to: https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/19401 As far as I know, In the langdale release, they have changed the binutils version from 2.38 -> 2.39, where 2.39 contains more checks.
> > > I have already prepared a patch for this, but I was going to include it in PR with support for OPIZ2 for Kernel 6x (still in progress, I'm still testing some things), but I can deliver it separately.
> > > Edit: They merged it to the 'integration' branch today. I'm not familiar with the development process of ATF but for me - if it was merged into the 'integration' branch it means that sooner or later it will be available in the mainline. The problem is that we still don't use the mainline ATF - except in the case of OPIZ2 available on 'kirkstone' branch and soon on master, so a patch is needed anyway unless we completely switch to mainline ATF.
> > 
> > 
> > Thanks, I've rolled back to `dunfell` and manually added support for OPC2, until a stable release become available in mainline.
> 
> You have OPC2 available on `kirkstone` branch and on this release it should work fine. Give it a shoot :)

Sure, Thanks :)

closed 2023-04-01 by nandra