linux-sunxi/meta-sunxi #370: Build fails for master branch and OPi Zero 2

jakub-kozlowicz, 2023-03-30:
Hi!

I have tried to build `core-image-base` image for the `Orange Pi Zero 2` board. For the poky `langdale` version I have encountered few errors.

First of them is error when patching kernel in version 6.1.9.
[kernerl_patch_error.txt](https://github.com/linux-sunxi/meta-sunxi/files/11112723/kernerl_patch_error.txt)

Second of them is error when compiling kernel modules and network drivers.
[kernel_modules_error.txt](https://github.com/linux-sunxi/meta-sunxi/files/11112712/kernel_modules_error.txt)

Also maybe it would be useful to create branches for newer releases of poky (Langdale, Mickledore).

nandra, 2023-03-30:
@pblxptr any ideas please? I didn't try yet to build opi zero 2 but will try and let you know. Thanks.

pblxptr, 2023-03-30:
> @pblxptr any ideas please? I didn't try yet to build opi zero 2 but will try and let you know. Thanks.

I will have a look at this

pblxptr, 2023-03-31:
Hmm it seems that something went wrong with file permissions in patches, @jakub-kozlowicz  can you have a look at this https://github.com/pblxptr/meta-sunxi/tree/opiz2-kernel6x-fix 
Could you check out this branch and see if it works? I tested it locally and it compiled properly, also booted it on board, and it seems to be working fine.

jakub-kozlowicz, 2023-03-31:
> Hmm it seems that something went wrong with file permissions in patches, @jakub-kozlowicz can you have a look at this https://github.com/pblxptr/meta-sunxi/tree/opiz2-kernel6x-fix
> Could you check [...]

Yeah, I will test it today and let you know :)

jakub-kozlowicz, 2023-03-31:
@pblxptr I have managed to successfully build image with langdale version of poky and your branch with fix. I've  encountered different problem. This image is not bootable. Red led on the board is on but the green led do not light on. I used a `dd` program to create SD card with the `sunxi-sdimg`, but I do not know if I should use some different switches.

pblxptr, 2023-03-31:
> @pblxptr I have managed to successfully build image with langdale version of poky and your branch with fix. I've encountered different problem. This image is not bootable. Red led on the board is on [...]

Hmm, are you sure that it's not booting? What's the output on a serial port? 
The red led is not automatically turned on when you plug in the power supply. It's managed by the Kernel, and enabled when the kernel parses the device tree and enables drivers. 

``

	leds {
		compatible = "gpio-leds";

		led-0 {
			function = LED_FUNCTION_POWER;
			color = <LED_COLOR_ID_RED>;
			gpios = <&pio 2 12 GPIO_ACTIVE_HIGH>; /* PC12 */
			default-state = "on";
		};

		led-1 {
			function = LED_FUNCTION_STATUS;
			color = <LED_COLOR_ID_GREEN>;
			gpios = <&pio 2 13 GPIO_ACTIVE_HIGH>; /* PC13 */
		};
	};

pblxptr, 2023-03-31:
@jakub-kozlowicz any update?

jakub-kozlowicz, 2023-03-31:
> @jakub-kozlowicz any update?

Yep, I've managed to boot and using serial port test Ethernet connectivity. Now I am working on wifi connection.

```
CORE_IMAGE_EXTRA_INSTALL:append = " ethtool networkmanager-nmcli kernel-modules linux-firmware uwe5622-firmware wpa-supplicant"
KERNEL_MODULE_AUTOLOAD += "sprdwl_ng"

DISTRO_FEATURES:append = " systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
```

This code with new version of poky requires meta-networking layer from meta-openembedded. Also there are two network-manager services installed (`/etc/init.d/network-manager`, `/lib/systemd/system/NetworkManager.service`). This cause conflicts and I cannot configure wifi connectivity. Another problem is that there is missing networkmanager-wifi plugin.

```
* network-manager.service - LSB: network connection manager
     Loaded: loaded (/etc/init.d/network-manager; generated)
     Active: active (running) since Fri 2023-03-31 14:32:26 UTC; 26min ago
       Docs: man:systemd-sysv-generator(8)
    Process: 216 ExecStart=/etc/init.d/network-manager start (code=exited, status=0/SUCCESS)
      Tasks: 3 (limit: 1153)
     Memory: 7.6M
     CGroup: /system.slice/network-manager.service
             `-218 /usr/sbin/NetworkManager

Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.1685] ifupdown: No dns-nameserver configured in /etc/network/interfaces
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.1707] ifupdown: guessed connection type (bnep0) = 802-3-ethernet
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.2024] dhcp: init: Using DHCP client 'internal'
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.2033] device (lo): carrier: link connected
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.2085] manager: (lo): new Generic device (/org/freedesktop/NetworkManager/Devices/1)
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.2178] manager: (wlan0): 'wifi' plugin not available; creating generic device
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.2307] manager: (wlan0): new Generic device (/org/freedesktop/NetworkManager/Devices/2)
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.2488] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/3)
Mar 31 14:32:28 orange-pi-zero2 NetworkManager[218]: <info>  [1680273148.2694] manager: startup complete
Mar 31 14:32:31 orange-pi-zero2 NetworkManager[218]: <info>  [1680273151.8718] device (eth0): carrier: link connected
```

```
x NetworkManager.service - Network Manager
     Loaded: loaded (/lib/systemd/system/NetworkManager.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-03-31 14:32:28 UTC; 26min ago
   Duration: 62ms
       Docs: man:NetworkManager(8)
    Process: 244 ExecStart=/usr/sbin/NetworkManager --no-daemon (code=exited, status=1/FAILURE)
   Main PID: 244 (code=exited, status=1/FAILURE)

Mar 31 14:32:28 orange-pi-zero2 systemd[1]: NetworkManager.service: Scheduled restart job, restart counter is at 5.
Mar 31 14:32:28 orange-pi-zero2 systemd[1]: Stopped Network Manager.
Mar 31 14:32:28 orange-pi-zero2 systemd[1]: NetworkManager.service: Start request repeated too quickly.
Mar 31 14:32:28 orange-pi-zero2 systemd[1]: NetworkManager.service: Failed with result 'exit-code'.
Mar 31 14:32:28 orange-pi-zero2 systemd[1]: Failed to start Network Manager.
```

Changes from your branch fixes problems with building image for langdale version of poky. I will post a minimal version of local.conf file with working wifi and ethernet when I finish it.

PR #371 (2023-04-01): Fix file permissions in patches for Orange Pi Zero 2 with Kernel 6x
This pull request fixes build problems that occur during patching the kernel. Fixes #370 issue.
closed 2023-04-01 by nandra