linux-sunxi/meta-sunxi #375: wic for sunxi64

ludiazv, 2023-05-03:
There is any reason that wic is not implemented for arm64? What would be the modifications to make in current wks and sunxi64.inc to support this mode?

Thanks,

nandra, 2023-05-03:
I think simply nobody update if to wic yet. Feel free to post pull request ;)

nandra, 2023-05-03:
I did quick check and tested in opi pc2 and it seems to boot:
```
diff --git a/conf/machine/include/sunxi64.inc b/conf/machine/include/sunxi64.inc
index b2df753..aeb5bb4 100644
--- a/conf/machine/include/sunxi64.inc
+++ b/conf/machine/include/sunxi64.inc
@@ -9,8 +9,7 @@ PREFERRED_PROVIDER_virtual/bootloader ?= "u-boot"
 
 KERNEL_IMAGETYPE ?= "Image"
 
-IMAGE_CLASSES += "sdcard_image-sunxi"
-IMAGE_FSTYPES += "ext4 tar.gz sunxi-sdimg"
+IMAGE_FSTYPES += "ext4 tar.gz wic.gz wic.bmap"
 
 MACHINE_EXTRA_RRECOMMENDS = "kernel-modules"
 
@@ -19,8 +18,13 @@ UBOOT_LOCALVERSION = "-g${@d.getVar('SRCPV', True).partition('+')[2][0:7]}"
 UBOOT_ENTRYPOINT ?= "0x40008000"
 UBOOT_LOADADDRESS ?= "0x400080OB00"
 
-#UBOOT_BINARY ?= "u-boot.itb"
-SPL_BINARY ?= "spl/sunxi-spl.bin"
+SPL_BINARY = "u-boot-sunxi-with-spl.bin"
 
 SERIAL_CONSOLE ?= "115200 ttyS0"
 MACHINE_FEATURES ?= "alsa apm keyboard rtc serial screen usbgadget usbhost vfat"
+
+WKS_FILES ?= "sunxi-sdcard-image.wks.in"
+WKS_FILE_DEPENDS ?= "virtual/kernel u-boot"
+SUNXI_BOOT_SPACE ?= "40"
+IMAGE_BOOT_FILES ?= "${KERNEL_IMAGETYPE} boot.scr ${KERNEL_DEVICETREE}"
```

nandra, 2023-05-04:
@ludiazv yes this was typo, please drop it. Let me know if board boots then I'll make pull request and merge it to master and kirkstone and dunfell branches. Thanks.

nandra, 2023-05-04:
@ludiazv sorry have fix for that. Will push changes shortly.

ludiazv, 2023-05-05:
Hi,
I came across this workarround as ```IMAGE_BOOT_FILES``` is limited to create directories.

```
...
MACHINE_FEATURES ?= "alsa apm keyboard rtc serial screen usbgadget usbhost vfat"

# location in deploy dir of the dbts
# arm64 dbts are under <vendor>/dts but is deployed under DEPLOYDIR
do_fix_device_tree_location() {

    for kdt in ${KERNEL_DEVICETREE}
    do
            local dbt_dir=$(dirname ${kdt})
            if [ "." != "${dbt_dir}" ] ; then
              local dbt=$(basename ${kdt})
              local dst=${DEPLOY_DIR_IMAGE}/${dbt_dir}/${dbt}
              if [ ! -f ${dst} ] ; then
               mkdir -p ${DEPLOY_DIR_IMAGE}/$dbt_dir
               ln -s ${DEPLOY_DIR_IMAGE}/${dbt} ${dst}
              fi
            fi
     done
}

addtask do_fix_device_tree_location after do_write_wks_template before do_image_wic

SUNXI_FEX_FILE ?= " "
SUNXI_BOOT_SPACE ?= "40"
IMAGE_BOOT_FILES ?= "${KERNEL_IMAGETYPE} boot.scr"
# ship fex file for sunxi kernel, devicetree file for mainline kernel
IMAGE_BOOT_FILES += "${@bb.utils.contains('PREFERRED_PROVIDER_virtual/kernel', 'linux-sunxi', '${SUNXI_FEX_FILE}', '${KERNEL_DEVICETREE}', d)}"
WKS_FILES ?= "sunxi-sdcard-image.wks.in"
WKS_FILE_DEPENDS ?= "virtual/kernel u-boot"
# in case of sunxi kernel, sunxi-board-fex must be built prior wic image assembly
WKS_FILE_DEPENDS += " ${@bb.utils.contains('PREFERRED_PROVIDER_virtual/kernel', 'linux-sunxi', 'sunxi-board-fex', '', d)}"

```
I send a PR if ok for you.

nandra, 2023-05-08:
Please send whole including wic so can merge in once. Thanks.

ludiazv, 2023-05-08:
Check-out the pull request to be merged in master.

I tested the wic in kirkstone an also works. (testing in a orange pi zero 2), So the change can also be backported to that branch.

closed 2023-05-11 by nandra