linux-sunxi/meta-sunxi #386: Should this generate an SD card image?

retpolanne, 2023-07-03:
Hello! Newbie Yocto user, sorry for this being more of a question than an issue. 

I'm trying to bake images for the Orange Pi One Plus. Baking went successfully and I tested the generated u-boot image and it loaded just fine. 

However, I can't see a file such as:

```
core-image-minimal-orange-pi-one-plus.wic.xz
```

Which I believe is the image I should dd to my sdcard that would contain u-boot, kernel and rootfs. 

The only wic.xz file I have is `core-image-minimal-orange-pi-one-plus.rootfs.wic.gz` which contains just the rootfs. 

My question: should meta-sunxi generate an sdcard image? Or do I have to do this manually?

retpolanne, 2023-07-03:
I added `IMAGE_FSTYPES += "wic.xz"` to my local.conf and it seems that, even though the xz has "rootfs" on its name, it contains u-boot as well. However, it doesn't boot: 

```
U-Boot SPL 2023.04-gfd4ed6b (Apr 03 2023 - 20:38:50 +0000)
DRAM: 1024 MiB
Trying to boot from MMC1
```

nandra, 2023-07-03:
Do you use master branch for building an image please? usually such log means something is wrong with ATF.

retpolanne, 2023-07-03:
Yes, I did use master. I don't really see ATF logs before u-boot though.

nandra, 2023-07-03:
I'm little puzzled why you have u-boot 2023.04. I build master for bananapi m64 and I have:
```
U-Boot SPL 2022.10-g4debc57 (Oct 03 2022 - 19:25:32 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.8(release):v2.8-dirty
NOTICE:  BL31: Built : 18:12:00, Nov 21 2022
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x20abde8, model: BananaPi-M64
```

retpolanne, 2023-07-03:
I'll try do do a clean build.

retpolanne, 2023-07-03:
Update - I did a clean build and everything is still on the master, but I still see u-boot as 2023.04.

nandra, 2023-07-04:
OK seems my poky master was not up to date. Checking now.

nandra, 2023-07-04:
Also work:
```
U-Boot SPL 2023.04-gfd4ed6b (Apr 03 2023 - 20:38:50 +0000)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: lts-v2.8.6(release):lts-v2.8.6-dirty
NOTICE:  BL31: Built : 17:57:15, Apr 21 2023
NOTICE:  BL31: Detected Allwinner A64/H64/R18 SoC (1689)
NOTICE:  BL31: Found U-Boot DTB at 0x20b1ce0, model: BananaPi-M64
```

Could it be that transition to mainline atf break something on H6 maybe?

nandra, 2023-07-04:
Can you please try to build kirkstone branch instead just to verify if it boots? Thanks.

retpolanne, 2023-07-04:
> Can you please try to build kirkstone branch instead just to verify if it boots? Thanks.

I also tried that, unfortunately it didn't boot :(

retpolanne, 2023-07-04:
This is the atf compilation step:

```
NOTE: trusted-firmware-a-2.8.6-r0 do_compile: make -j 12 LD=aarch64-poky-linux-ld CC=aarch64-poky-linux-gcc V=1 E=0 BUILD_BASE=/home/retpolanne/Dev/orange-pi-one-plus-image/orange-pi-one-plus/tmp/work/orange_pi_one_plus-poky-linux/trusted-firmware-a/2.8.6-r0/build PLAT=sun50i_a64 OPENSSL_DIR=/home/retpolanne/Dev/orange-pi-one-plus-image/orange-pi-one-plus/tmp/work/orange_pi_one_plus-poky-linux/trusted-firmware-a/2.8.6-r0/recipe-sysroot-native//usr HOSTCC=gcc  RUNTIME_SYSROOT=/home/retpolanne/Dev/orange-pi-one-plus-image/orange-pi-one-plus/tmp/work/orange_pi_one_plus-poky-linux/trusted-firmware-a/2.8.6-r0/recipe-sysroot -C /home/retpolanne/Dev/orange-pi-one-plus-image/orange-pi-one-plus/tmp/work/orange_pi_one_plus-poky-linux/trusted-firmware-a/2.8.6-r0/git bl31
```

This is the the PLAT:

`PLAT=sun50i_a64`

Shouldn't this be `sun50i_h6` for H6?

retpolanne, 2023-07-04:
Holy Cow!!! It booted: 

```
diff --git a/recipes-bsp/trusted-firmware-a/trusted-firmware-a_%.bbappend b/recipes-bsp/trusted-firmware-a/trusted-firmware-a_%.bbappend
index fc0d715..99655d4 100644
--- a/recipes-bsp/trusted-firmware-a/trusted-firmware-a_%.bbappend
+++ b/recipes-bsp/trusted-firmware-a/trusted-firmware-a_%.bbappend
@@ -1,6 +1,6 @@
 COMPATIBLE_MACHINE = "(sun50i|sun50i-h616)"
 
-TFA_PLATFORM:sun50i = "sun50i_a64"
+TFA_PLATFORM:sun50i = "sun50i_h6"
 TFA_PLATFORM:sun50i-h616 = "sun50i_h616"
 
 TFA_BUILD_TARGET = "bl31"
 ```

nandra, 2023-07-04:
Ok nice. Thanks I'll make changes to fix it in master + kirkstone

nandra, 2023-07-04:
Sure thanks.

retpolanne, 2023-07-04:
Hey @nandra ! Thanks for that :). `atf-fixes` booted just fine!

nandra, 2023-07-07:
Pushed to master. Closing.

closed 2023-07-07 by nandra