linux-sunxi/meta-sunxi #387: Ethernet not working on Orange Pi One Plus (H6)

retpolanne, 2023-07-04:
Hello! I wanted to use this issue to document what we need to enable Ethernet on the Orange Pi One Plus. 

It seems that the Orange Pi One Plus doesn't have upstream support for ethernet: 

```
U-Boot 2023.04-gfd4ed6b (Apr 03 2023 - 20:38:50 +0000) Allwinner Technology

CPU:   Allwinner H6 (SUN50I)
Model: OrangePi One Plus
DRAM:  2 GiB
Core:  53 devices, 16 uclasses, devicetree: separate
WDT:   Not starting watchdog@7020400
MMC:   mmc@4020000: 0
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... 
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
starting USB...
Bus usb@5101000: USB EHCI 1.00
Bus usb@5101400: USB OHCI 1.0
Bus usb@5311000: USB EHCI 1.00
Bus usb@5311400: USB OHCI 1.0
scanning bus usb@5101000 for devices... 1 USB Device(s) found
scanning bus usb@5101400 for devices... 1 USB Device(s) found
scanning bus usb@5311000 for devices... 1 USB Device(s) found
scanning bus usb@5311400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
=> dhcp
No ethernet found.
```

I made a hacky patch (attached to this issue) and added it to sunxi bbappends. 
[0003-sunxi-H6-Enable-Ethernet-on-Orange-Pi-One-Plus.patch](https://github.com/linux-sunxi/meta-sunxi/files/11951460/0003-sunxi-H6-Enable-Ethernet-on-Orange-Pi-One-Plus.patch)

```
=> dhcp
sun8i_emac_eth_start: Timeout
```

These are a few links I'm using for reference: 

https://patchwork.ozlabs.org/project/uboot/patch/20200507231052.23706-3-samuel@sholland.org/
https://patchwork.kernel.org/project/linux-arm-kernel/patch/20180722053955.25266-6-icenowy@aosc.io/
https://patchwork.ozlabs.org/project/uboot/patch/20180214230216.2001-2-kevans@FreeBSD.org/

retpolanne, 2023-07-11:
Sent a patch fixing it :)

https://lore.kernel.org/u-boot/20230711003957.658805-2-retpolanne@posteo.net/T/#u

nandra, 2023-07-11:
ok nice, can you post PR with this patch to this meta layer please? Thanks.

retpolanne, 2023-07-11:
Sure! Will do

retpolanne, 2023-07-12:
Patch sent to upstream and added to bbappend. Closing this ;)

closed 2023-07-12 by retpolanne