linux-sunxi/meta-sunxi #392: Upcoming mainline kernels support

buldo, 2023-10-03:
Hello.  
This is not real issue right now but I want to highlight it.

I tried to add recipe for 6.5.5 kernel. But in this kernel version( as I understand) dts patches was changed. I got error like "***.dtb target not found".

pblxptr, 2023-10-03:
I don't understand what's the point of this post.

buldo, 2023-10-03:
As I understand machine configuration files will require dfferent `KERNEL_DEVICETREE` for kernels >=6.5.5.
I wrote this issue to highlight potential future problem and discuss it

nandra, 2023-10-03:
I didn't check by myself but can be related to that: https://lists.yoctoproject.org/g/yocto/topic/meta_rockchip_patch/101728482?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,101728482,previd%3D1696346558264871138,nextid%3D1696236622653823727&previd=1696346558264871138&nextid=1696236622653823727

buldo, 2023-10-03:
@nandra yes.  
What to do with this changes? For example, if someone wants to add new kernel to meta-sunxi, how to made machine configs that will works with old and new kernels?

nandra, 2023-10-03:
Well it's about sanitizing paths for 32 bit CPU's. 64bit CPU already use that scheme for dtb so we can copy some bits from 64 bit for new kernels. I'll look into it and post PR. Thanks.

ludiazv, 2023-10-04:
@buldo  Check out this fix that I implemented for 64bit that could be useful for 32bit.

https://github.com/linux-sunxi/meta-sunxi/blob/6c6ef34b6980421237b6e9443bc959e06bfc7e89/conf/machine/include/sunxi64.inc#L27C1-L42C2

A conditional based on Kernel version & target arch to prepend "allwinner/" to ```KERNEL_DEVICE_TREE``` in linux-mainline.inc can patch. This will avoid to modify the machine files and make them work with new and old mainline kernel.

nandra, 2023-10-24:
@ludiazv @buldo I've prepared PR for 6.5 kernel handling. Compile tested only. Please comment: https://github.com/linux-sunxi/meta-sunxi/pull/395

buldo, 2023-11-01:
@nandra I tested it on Pi Lite 1G. core-image-full-cmdline successfuly loads

nandra, 2023-11-10:
Merged changes to master + set nanbield release compatible.

closed 2023-11-10 by nandra