linux-sunxi/meta-sunxi #398: orange-pi-zero2 build no longer works

electricworry, 2023-11-11:
Attempting to build core-image-minimal for the orange-pi-zero2 but it now fails on master branch. (Possibly since the update to target linux-mainline 6.5.11 or moving to poky nanbield.)

```
| make[3]: *** No rule to make target 'arch/arm64/boot/dts/allwinner/allwinner/sun50i-h616-orangepi-zero2.dtb'.  Stop.
| make[2]: *** [/home/electricworry/projects/orangepizero2/build/tmp/work-shared/orange-pi-zero2/kernel-source/scripts/Makefile.build:480: arch/arm64/boot/dts/allwinner] Error 2
| make[1]: *** [/home/electricworry/projects/orangepizero2/build/tmp/work-shared/orange-pi-zero2/kernel-source/Makefile:1469: allwinner/allwinner/sun50i-h616-orangepi-zero2.dtb] Error 2
| make: *** [/home/electricworry/projects/orangepizero2/build/tmp/work-shared/orange-pi-zero2/kernel-source/Makefile:234: __sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/electricworry/projects/orangepizero2/layers/meta-sunxi/recipes-kernel/linux/linux-mainline_6.5.11.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 859 tasks of which 845 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/electricworry/projects/orangepizero2/layers/meta-sunxi/recipes-kernel/linux/linux-mainline_6.5.11.bb:do_compile
```

sun50i-h616-orangepi-zero2.dts is in the kernel source. Any idea how to resolve this?

electricworry, 2023-11-11:
A-ha, should the following line now have "allwinner/" removed?

https://github.com/linux-sunxi/meta-sunxi/blob/40c6d521230f24094cd189bb645284da2fcb2804/conf/machine/orange-pi-zero2.conf#L7

I see this comment in recipes-kernel/linux/linux-mainline.inc:

```
# kernel 6.5 have updated dts path add 'allwinner' prefix automatically
```

nandra, 2023-11-13:
Let me try to build locally and push fix to master. Thanks for reporting.

nandra, 2023-11-13:
@electricworry fix pushed to master branch. Compile tested only.

nandra, 2023-11-13:
Please verify and let me know. Thanks.

electricworry, 2023-11-13:
All resolved now, thank you!

closed 2023-11-13 by electricworry