linux-sunxi/meta-sunxi #399: [OrangePi PC2] Wrong place for dtb in boot partition

javad123javad, 2023-11-22:
Hi,
I've made a minimal image from master branch. After burning the `wic` image on the sdcard, The device tree file is located in <boot partition> while the u-boot is trying to load it from `allwiner/<device tree>.dtb` . How should I solve this problem? 
```
Found U-Boot script /boot.scr
537 bytes read in 3 ms (174.8 KiB/s)
## Executing script at 4fc00000
Failed to load 'allwinner/sun50i-h5-orangepi-pc2.dtb'

```

nandra, 2023-11-22:
@javad123javad which commit did you use from master please? I've pushed recently fix for this issue. Thanks.

javad123javad, 2023-11-22:
> @javad123javad which commit did you use from master please? I've pushed recently fix for this issue. Thanks.

Thanks, `git pull` solved the problem. My repo was old.

closed 2023-11-22 by javad123javad