linux-sunxi/meta-sunxi #402: nanopi-r1 machine fails to build in u-boot for kirkstone (missing dts)

asac, 2024-04-24:
Trying to build nanopi-r1 machine on kirkstone branch i have a missing dts in u-boot 2022-01 that is needed for the defconfig that was added by this meta layer.

```
zeropi.dtb.d.dtc.tmp > arch/arm/dts/.sun8i-h3-zeropi.dtb.d
| test -e arch/arm/dts/sun8i-h3-nanopi-r1.dtb || (						\
| echo >&2;							\
| echo >&2 "Device Tree Source (arch/arm/dts/sun8i-h3-nanopi-r1.dtb) is not correctly specified.";	\
| echo >&2 "Please define 'CONFIG_DEFAULT_DEVICE_TREE'";		\
| echo >&2 "or build with 'DEVICE_TREE=<device_tree>' argument";	\
| echo >&2;							\
| /bin/false)
| 
| Device Tree Source (arch/arm/dts/sun8i-h3-nanopi-r1.dtb) is not correctly specified.
| Please define 'CONFIG_DEFAULT_DEVICE_TREE'
| or build with 'DEVICE_TREE=<device_tree>' argument
| 
| make[2]: *** [/home/panta/yocto/panta/meta/meta-pantavisor/build/tmp-kirkstone/work/nanopi_r1-poky-linux-gnueabi/u-boot/1_2022.01-r0/git/dts/Makefile:34: arch/arm/dts/sun8i-h3-nanopi-r1.dtb] Error 1
| make[1]: *** [/home/panta/yocto/panta/meta/meta-pantavisor/build/tmp-kirkstone/work/nanopi_r1-poky-linux-gnueabi/u-boot/1_2022.01-r0/git/Makefile:1134: dts/dt.dtb] Error 2
| make[1]: Leaving directory '/home/panta/yocto/panta/meta/meta-pantavisor/build/tmp-kirkstone/work/nanopi_r1-poky-linux-gnueabi/u-boot/1_2022.01-r0/build'
| make: Leaving directory '/home/panta/yocto/panta/meta/meta-pantavisor/build/tmp-kirkstone/work/nanopi_r1-poky-linux-gnueabi/u-boot/1_2022.01-r0/git'
| make: *** [Makefile:177: sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/panta/yocto/panta/meta/meta-pantavisor/build/../poky/meta/recipes-bsp/u-boot/u-boot_2022.01.bb:do_compile) failed with exit code '1'
```

nandra, 2024-04-24:
@asac thanks for there post. I'll check it.

nandra, 2024-04-24:
@asac please check this PR #404 . Also if you have board please try if it boots. Thanks.

asac, 2024-04-24:
@nandra thanks. the PR builds successfully. I don't have the device (yet). Found this issue while integrating sunxi BSPs for meta-pantavisor.

But let me get one and report back...

nandra, 2024-04-25:
@asac ok thanks. I'll close this one and if something won't work feel free to open new one ;). Thanks.

closed 2024-04-25 by nandra

asac, 2024-04-25:
Hey @nandra ... today got R1 device and it boots. Havent gone deeper into drivers etc. but thanks for the quick fix on  this. well done...

```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.5.11 (oe-user@oe-host) (arm-poky-linux-gnueabi-gcc (GCC) 13.2.0, GNU ld (GNU Binutils) 2.42.0.20240216) #1 SMP Wed Nov  8 13:09:07 UTC 2023
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: FriendlyARM NanoPi R1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000006fffffff]
[    0.000000]   HighMem  [mem 0x0000000070000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 13 pages/cpu s23060 r8192 d21996 u53248
[    0.000000] Kernel command line: earlyprintk root=/dev/ram rootfstype=ramfs rdinit=/usr/bin/pantavisor pv_try= pv_rev=0 panic=2 pv_quickboot   
[    0.000000] Unknown kernel command line parameters "pv_quickboot pv_try= pv_rev=0", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 260416
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 1017420K/1048576K available (9216K kernel code, 894K rwdata, 1972K rodata, 1024K init, 286K bss, 31156K reserved, 0K cma-reserved, 262144K highmem)
```

asac, 2024-04-25:
oh that wasnt kirkstone... let me spin that ...

asac, 2024-04-25:
ok seems that the uboot also boots from kirkstone now... thx