linux-sunxi/meta-sunxi #407: Legacy sunxi Kernel/U-Boot

berkutta, 2024-05-03:
I'm trying to get this meta layer running for an A20 Board. Unfortunately, the Board uses NAND Flash. Thus I need to use the legacy kernel.

Is the legacy kernel/u-boot still supported for scarthgap?

I have followed #legacy-sunxi-kernel--u-boot in the readme and added the four lines into my local.conf. But then it refuses to build an image (core-image-minimal):
```
$ bitbake core-image-minimal -k
Loading cache: 100% |##############################################################################################################################################################################| Time: 0:00:00
Loaded 4402 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'u-boot' (but /yocto/poky/meta/recipes-core/images/core-image-minimal.bb DEPENDS on or otherwise requires it)
u-boot was skipped: PREFERRED_PROVIDER_virtual/bootloader set to u-boot-sunxi, not u-boot
ERROR: Nothing RPROVIDES 'core-image-minimal'
No eligible RPROVIDERs exist for 'core-image-minimal'
NOTE: Runtime target 'core-image-minimal' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['core-image-minimal']

Build Configuration:
BB_VERSION           = "2.8.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "bananapi"
DISTRO               = "poky"
DISTRO_VERSION       = "5.0"
TUNE_FEATURES        = "arm vfp cortexa7 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "scarthgap:3e10e2afba7711a6e7a42467339e737ea3196285"
meta-oe              
meta-python          = "scarthgap:a72010b414ee3d73888ac9cb4e310e8f05e13aea"
meta-arm             
meta-arm-toolchain   = "master:8aa8a1f17f5b64bc691544f989f04fc83df98adb"
meta-sunxi           = "master:c475a1851a877bf700d6d9748be7336b2eadd3b2"

ERROR: All buildable tasks have been run but the build is incomplete (--continue mode). Errors for the tasks that failed will have been printed above.                                             | ETA:  0:00:00
ERROR: Command execution failed: 1

Summary: There were 4 ERROR messages, returning a non-zero exit code.
```

I'm trying to get this meta layer running on an A20 Board. Unfortunately, the Board uses NAND Flash. Thus, I need to use the legacy kernel.

Is the legacy kernel/u-boot still supported for scarthgap?

I have followed #legacy-sunxi-kernel--u-boot in the README and added the four lines into my local.conf. But then it refuses to build an image (core-image-minimal):
```
|   CC      fs/proc/proc_sysctl.o
|   CC      fs/proc/kmsg.o
|   CC      fs/proc/page.o
| In function 'free_note_info',
|     inlined from 'elf_core_dump' at /yocto/build/tmp/work-shared/bananapi/kernel-source/fs/binfmt_elf.c:2087:2:
| /yocto/build/tmp/work-shared/bananapi/kernel-source/fs/binfmt_elf.c:1620:27: warning: 'info.psinfo.data' may be used uninitialized [-Wmaybe-uninitialized]
|  1620 |         kfree(info->psinfo.data);
|       |               ~~~~~~~~~~~~^~~~~
| /yocto/build/tmp/work-shared/bananapi/kernel-source/fs/binfmt_elf.c: In function 'elf_core_dump':
| /yocto/build/tmp/work-shared/bananapi/kernel-source/fs/binfmt_elf.c:1911:30: note: 'info' declared here
|  1911 |         struct elf_note_info info;
|       |                              ^~~~
|   LD      fs/proc/proc.o
|   LD      fs/proc/built-in.o
|   LD      fs/built-in.o
| make[1]: *** [Makefile:130: sub-make] Error 2
| make: *** [Makefile:26: all] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
ERROR: Task (/yocto/build/../meta-sunxi/recipes-kernel/linux/linux-sunxi_3.4.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3639 tasks of which 3638 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /yocto/build/../meta-sunxi/recipes-kernel/linux/linux-sunxi_3.4.bb:do_compile
Summary: There were 3 ERROR messages, returning a non-zero exit code.
```

Do I need to use a specially old compiler or something else to get these legacy components to build?

nandra, 2024-05-07:
@berkutta I've never used sunxi kernel/u-boot but I think some time ago there was some patch for update which is merged. I'll check it and get back to you. Thanks.

nandra, 2024-05-13:
@berkutta I even cannot download kernel :(

berkutta, 2024-05-13:
@nandra Thanks for checking! What error are you getting during download?
I have tried it again right now. do_fetch works for me for `linux-sunxi-3.4.104-r1`.

nandra, 2024-05-13:
It works for you because you have it already cloned. I'll try tomorrow again.

nandra, 2024-08-16:
@berkutta do we still need this issue or you was able to resolve it pls? Thanks.

berkutta, 2024-08-16:
I wasn't able to resolve this. Unsure if building the Legacy Sunxi Kernel is even possible in a current Yocto.