linux-sunxi/meta-sunxi #410: olinuxino-a20lime2-emmc: pya20 SRC_URI entry mapping.h: file could not be found

ardavast, 2024-05-12:
I got an error while trying to compile for MACHINE = "olinuxino-a20lime2-emmc"

I was able to fix it by doing:
```
cd recipes-devtools/python/files
ln -s olinuxino-a20lime2 olinuxino-a20lime2-emmc
```

The error was:
```
Loading cache: 100%
Loaded 0 entries from dependency cache.                                                                                                                                                                                                                                           
WARNING: /home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20_0.2.12.bb: distutils-common-base.bbclass is deprecated, please use setuptools3-base.bbclass instead                                                                                            
WARNING: /home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20_0.2.12.bb: distutils3-base.bbclass is deprecated, please use setuptools3-base.bbclass instead                                                                                                  
WARNING: /home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20_0.2.12.bb: distutils3.bbclass is deprecated, please use setuptools3.bbclass instead                                                                                                            
ERROR: /home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20_0.2.12.bb: Unable to get checksum for pya20 SRC_URI entry mapping.h: file could not be found                                                                                                     
The following paths were searched:                                                                                                       
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/poky/mapping.h                                          
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/poky/mapping.h                                                 
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/poky/mapping.h                                                                                                                                                                                          
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/use-mailine-graphics/mapping.h                          
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/use-mailine-graphics/mapping.h                                 
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/use-mailine-graphics/mapping.h                                 
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/olinuxino-a20lime2-emmc/mapping.h                                                                                                                                                                
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/olinuxino-a20lime2-emmc/mapping.h                                                                                                                                                                       
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/olinuxino-a20lime2-emmc/mapping.h                                                                                                                                                                       
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/sun7i/mapping.h                                                                                                                                                                                  
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/sun7i/mapping.h                                                                                                                                                                                         
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/sun7i/mapping.h                                                                                                                                                                                         
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/sunxi/mapping.h                                         
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/sunxi/mapping.h                                                
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/sunxi/mapping.h                                                
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/armv7ve/mapping.h                                       
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/armv7ve/mapping.h                                                                                                                                                                                       
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/armv7ve/mapping.h                                                                                                                                                                                       
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/arm/mapping.h                                                                                                                                                                                    
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/arm/mapping.h                                                                                                                                                                                           
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/arm/mapping.h                                                                                                                                                                                           
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20-0.2.12/mapping.h                                                                                                                                                                                        
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/pya20/mapping.h                                                                                                                                                                                               
/home/ardavast/src/yocto/poky/../meta-sunxi/recipes-devtools/python/files/mapping.h                                                      
ERROR: Parsing halted due to errors, see error messages above
```

nandra, 2024-05-13:
@ardavast thanks for the report and fix. I've pushed it to master + scarthgap branches

ardavast, 2024-05-13:
Thanks! But it seems to me that the symlink commited to the repo is wrong, can you check it again?

nandra, 2024-05-13:
https://github.com/linux-sunxi/meta-sunxi/commit/322843f5766bf9899e760c10dbe5aca9e239c451