linux-sunxi/meta-sunxi #414: Bootable SPI flash

bartiszosti, 2024-08-11:
I'd like to place whole image on the external SPI flash and to do so I've done some changes.
According to the official documentation I've added to the u-boot config these lines:
```
CONFIG_SPL_SPI_SUNXI=y
CONFIG_SPI=y
CONFIG_SUN4I_SPI=y
CONFIG_CMD_SF=y
CONFIG_CMD_SSPI=y
CONFIG_DM_SPI=y
CONFIG_DM_SPI_FLASH=y
CONFIG_SPI_FLASH=y
CONFIG_SPI_FLASH_MACRONIX=y
CONFIG_SPI_FLASH_WINBOND=y
```
I've also modified the .dts file to make partitions on the spi0 flash:
```
 &spi0 {
	pinctrl-names = "default";
	pinctrl-0 = <&spi0_pc_pins>;
	status = "okay";

	flash@0 {
		compatible = "winbond,w25q128", "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <40000000>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "u-boot";
				reg = <0x000000 0x80000>;
			};

			partition@100000 {
				label = "boot";
				reg = <0x80000 0x680000>;
			};

			partition@510000 {
				label = "rootfs";
				reg = <0x680000 0x1000000>;
			};
		};
	};
};
```
I've also modified the boot.cmd
```
# Default to (primary) SD
rootdev=mtdblk0p2
if itest.b *0x28 == 0x02 ; then
	# U-Boot loaded from eMMC or secondary SD so use it for rootfs too
	echo "U-boot loaded from eMMC or secondary SD"
	rootdev=mtdblk1p2
fi
echo "aaaa ${console} ${rootdev} ${extra} ${fdt_addr_r} ${fdtfile} ${kernel_addr_r}"
setenv bootargs console=${console} console=tty1 root=/dev/${rootdev} rootwait panic=10 ${extra}
load mtd 0:1 ${fdt_addr_r} ${fdtfile} || load mtd 0:1 ${fdt_addr_r} boot/allwinner/${fdtfile}
load mtd 0:1 ${kernel_addr_r} zImage || load mtd 0:1 ${kernel_addr_r} boot/zImage || load mtd 0:1 ${kernel_addr_r} uImage || load mtd 0:1 ${kernel_addr_r} boot/uImage
bootz ${kernel_addr_r} - ${fdt_addr_r} || bootm ${kernel_addr_r} - ${fdt_addr_r}
```
Finally, I ve chenged the .wks.in file as below:
```
part u-boot --source rawcopy --sourceparams="file=${SPL_BINARY}" --no-table --align 8 --fixed-size 512k
part /boot --source bootimg-partition --fstype=vfat --label boot --active --align 2048 --fixed-size 6M
part / --source rootfs --fstype=ext4 --align 2048 --fixed-size 9M
```
At the end I got an .wic file, but I need a .bin file how to transform an .wic image to different format?
What is the UBOOT_ENTRYPOINT variable?
Did I do all necassary changes?

nandra, 2024-08-14:
Wic image can be used for block devices only. If you want to have image generated for spi flash then you need format which is valid for spi flash which can be jffs2 e.g. and this you can then flashed to spi flash. I never tried that so you need to try. Thanks.

bartiszosti, 2024-08-14:
Hello,
thanks for you answer. It was my bad, now I know how to use the SPI flash, but I met next issue.
I'd like to partition the SPI flash, but u-boot doesn't see any MTD device.
What configuration I need to enable it? Do you know?

nandra, 2024-08-14:
Hi, please check if you have options enabled: https://github.com/u-boot/u-boot/blob/master/drivers/mtd/Kconfig

closed 2024-11-11 by nandra