linux-sunxi/meta-sunxi #421: bananapi-m2berry support

vinsfortunato, 2024-10-20:
I'm trying to build a yocto image (scarthgap) for the Bananapi M2 Berry Board.

I've added the following machine configuration file (bananapi-m2berry.conf):
```
#@TYPE: Machine
#@NAME: bananapi-m2berry
#@DESCRIPTION: Machine configuration for the Banana Pi M2 Berry, base on Allwinner V40 CPU

require conf/machine/include/sun8i.inc

KERNEL_DEVICETREE = "sun8i-v40-bananapi-m2-berry.dtb"
UBOOT_MACHINE = "bananapi_m2_berry_defconfig"
```

This board should be supported by mainline kernel/u-boot since those DTS and defconf files are present.

I've built the image using `bitbake core-image-base`

The image was flashed to an SD Card with `dd`. There is no eMMC on board. Just the SD card.

However when the kernel is booting it is getting stuck at:
 `[    1.382638] Waiting for root device /dev/mmcblk0p2...`

The full boot log (The strange <0x83><0x08><0x9a>... sequence is due to my serial-to-usb device not working properly):
```
<break>

U-Bo<0x83><0x08><0x9a>A1<0x81>2024.01-g (Jan 08 2024 - 15:37:48 +0000)
DRAM: 1024 MiB
Trying to boot from MMC1

U,Boot 2024.01-g (Jan 08 2024 - 15:37:48 +0000) Allwinner Technology

CPU:   All<0x07>inner R40 (SUN8I 1701)
Model: Banana Pi M2 Berry
DRAM:  1 GiB
Core:  60 deWk<0x95><0xcd><0xb1> 23 uclasses, devicetree: separate
WDT:   Not starting watchdog@0c20c90
MMC:   mmc@0c0f000: 0, mmc@0c10000: 1
Loading Dnvironment from FAT... Unable to read "uboot.env" from mmc0:1... 
In:    serial@0c28000
Out:   serial@0c28000
Err:   serial@0c28000
Net:   eth0: ethernet@0c50000
starting USB...
Bus usb@0c19000: USB EHCI 1.00
scanning bus usb@0c19000 for devices... 2 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  2 <0x08><0x08><0x08> 1 <0x08><0x08><0x08> 0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot<0x9a><0x8d><0xc9><0xa5><0xc1><0xd1><0x81>/boot.scr
724 bytes read in 1 ms (707 KiB/s)
## Executing script at 43100000
25218 bytes read in 3 ms (8 MiB/s)
Failed to load 'zImage'
Failed to load 'boot/zImage'
5405760 bytes read in 224 ms (23 MiB/s)
zimage: Bad magic!
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-6.6.28
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    5405696 Bytes = 5.2 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
Working FDT set to 43000000
[... 233 lines cut ...]
+
+ 		osc24M_32k: clk@1 {
+			#clock-cells = <0>;
+			compatible = "fixed-factor-clock";
+			clock-div = <732>;
+			clock-mult = <1>;
+			clocks = <&osc24M>;
+			clock-output-names = "osc24M_32k";
+		};
+
+		clk_out_a: clk@01c201f0 {
+			#clock-cells = <0>;
+			compatible = "allwinner,sun7i-a20-out-clk";
+			reg = <0x01c201f0 0x4>;
+			clocks = <&osc24M_32k>, <&osc32k>, <&osc24M>;
+			clock-output-names = "clk_out_a";
+		};
 	};
 
 	cpus {
-- 
2.44.1
```

Didn't solve anything. I don't know what to try next. Unfortunately I'm also pretty new to yocto/linux.

vinsfortunato, 2024-10-25:
I solved the issue by adding the following configs to `recipes-kernel/linux/linux-mainline/arm/defconfig`:
```
CONFIG_MFD_AXP20X=y
CONFIG_MFD_AXP20X_I2C=y
CONFIG_MFD_AXP20X_RSB=y
CONFIG_REGULATOR_AXP20X=y
CONFIG_LEDS_AXP20X=m
CONFIG_AXP20X_ADC=m
CONFIG_INPUT_AXP20X_PEK=y
CONFIG_BATTERY_AXP20X=m
CONFIG_AXP20X_POWER=m
```

I discovered the issue by modifying the kernel to print the reason behind deferred probe pending and got this on boot:

```
[   12.046144] platform 1c0f000.mmc: deferred probe pending: platform: wait for supplier /soc/i2c@0c2ac00/pmic@04/regulators/dcdc1
[   12.057646] platform 1c10000.mmc: deferred probe pending: platform: wait for supplier /soc/i2c@0c2ac00/pmic@04/regulators/dldo1
```

And looking into the DTS I noticed that this required the AXP20X driver for I2C.

```
&i2c0 {
	status = "okay";

	axp22x: pmic@34 {
		compatible = "x-powers,axp221";
		reg = <0x34>;
		interrupt-parent = <&nmi_intc>;
		interrupts = <0 IRQ_TYPE_LEVEL_LOW>;
	};
};
```

The kernel boots. But now the HDMI is not working.

nandra, 2024-10-26:
:+1: thanks for sharing. If you resolve it you can post pull request and we can add this machine to supported ones by yocto. Thansk.

closed 2024-11-11 by nandra