linux-sunxi/meta-sunxi #427: HOW to build a yacto for customized board

deadly16soma, 2024-12-10:
Hi 
i am new to this yacto build. i want to build a yacto for my customized (components are changed ) board. what is the steps to take at starting to build.(is firmware is necessar if yes how to build it). should i create a device tree 
@leon-anavi

nandra, 2024-12-10:
@deadly16soma this is very broad question. Please see to some links on google which shows tons of manuals how to do that: https://stackoverflow.com/questions/50172362/to-build-custom-bsp-layer-using-yocto-for-a-bare-metal-board
https://community.nxp.com/t5/i-MX-Processors/how-to-add-custom-board-using-Yocto/m-p/345756

closed 2024-12-10 by nandra

leon-anavi, 2024-12-10:
@deadly16soma this a long and a complex topic and depends on the technical details. In a nutshell you need a working bootloader like U-Boot and a Linux kernel, both with proper defconfig and device tree (dts) files. After that you can integrated them in a Yocto/OpenEmbedded BSP (Board Support Package) layer, for example if the board is with Allwinner (sunXi) SoC you can contribute it to meta-sunxi. Here a couple of my recent contributions for adding Merrii A80 Optimus and Cubieboard 4 as an example:
https://github.com/linux-sunxi/meta-sunxi/pull/424
https://github.com/linux-sunxi/meta-sunxi/pull/425