linux-sunxi/meta-sunxi #433: Invalid hex number in sunxi64.inc

rm5248, 2025-01-03:
The following line contains an invalid hex number:
https://github.com/linux-sunxi/meta-sunxi/blob/a00dc47b24b6be35f1f6c76b3db31d0a50e56cd4/conf/machine/include/sunxi64.inc#L20

I'm not sure what it is supposed to be; I changed it from `0x400080OB00` to `0x40008000` on my build.

I only noticed this once I tried to create a FIT image of the kernel, it was unable to create the image due to the invalid hex in the DTS file that creates the FIT image.

nandra, 2025-01-03:
@rm5248 good catch. Thanks I'll take a look and let you know.

nandra, 2025-01-07:
@rm5248 should be fixed in #435 . Can you please check? Thanks.

rm5248, 2025-01-07:
I checked locally, it looks good to me.

nandra, 2025-01-07:
@rm5248 thanks. I've merged #435 to master branch.

closed 2025-01-07 by nandra