linux-sunxi/meta-sunxi #436: Panfrost freezes the boot process on OrangePi Zero 2

iliaalizadee, 2025-01-14:
Hi. When I enable the Panfrost driver using "use-mailine-graphics" the boot process halts and on-board leds stop blinking after these 2 messages:
```
panfrost 1800000.gpu: clock rate = 432000000
panfrost 1800000.gpu: bus_clock rate = 200000000
```
My target is the mainline kernel "6.6.28". No additional patches are involved and the image is built using core-image-base.

nandra, 2025-01-15:
Sorry I don't have board to test it locally. Maybe @ludiazv can have an idea?

iliaalizadee, 2025-01-17:
I can confirm that either the panfrost driver or gpu specification in device tree is problematic. I tested on 6.12.9 kernel with panfrost blacklisted and it worked fine. loading panfrost using modprobe freezes the system again.

iliaalizadee, 2025-01-17:
I made a buildsystem using Buildroot and tested the same case using 6.12.9 kernel. still the same problem. Tried blacklisting panfrost and manually loading it using modprobe, no difference. I thought the device tree specifications might be problematic, so I used my device tree with the official Debian Bookworm image (6.1.31 kernel) but panfrost worked fine so device tree is fine. Maybe sunxi patched something in the kernel or newer kernels are problematic? I don't know.

nandra, 2025-01-17:
@iliaalizadee you can try to ask on linux mailing list about that issue and maybe try latest 6.13-rc7 (I can make a branch if you want) to verify and then report on linux mailing list to get some insights. Thanks.

closed 2025-03-17 by nandra