linux-sunxi/meta-sunxi #438: master branch build is broken due to mesa

noodlefighter, 2025-02-18:
Hello, I'm using the master branch of linux-sunxi, and the dependent meta-poky branch is "styhead". I'm getting an error during the build:

```
ERROR: No recipes in default available for:
  /home/vagrant/h3-yocto/gd-yocto-aw-t113/build/../sources/meta-sunxi/recipes-graphics/mesa/mesa-gl_%.bbappend
  /home/vagrant/h3-yocto/gd-yocto-aw-t113/build/../sources/meta-sunxi/recipes-graphics/mesa/mesa_%.bbappend

```

Upon checking, I found that in the styhead branch of meta-poky, mesa_24.0.7.bb has been renamed to mesa.bb, and mesa-gl_24.0.7.bb has also been renamed to mesa-gl.bb.

The solution is to change mesa-gl_%.bbappend and mesa_%.bbappend to mesa-gl.bbappend and mesa.bbappend.

nandra, 2025-02-19:
@noodlefighter thanks for the report. I'm off this week but you can send pull request with that change and check it works so I can apply it. Thanks.

okaestne, 2025-03-09:
@noodlefighter are you sure you have the latest master? This should have been fixed 5 months ago with https://github.com/linux-sunxi/meta-sunxi/commit/9f2f3695498182ffbf64ca54ef78a9592dade5f8

closed 2025-03-17 by nandra