linux-sunxi/meta-sunxi #442: Question: Use upstream defconfigs and mainline graphics drivers by default?

okaestne, 2025-03-24:
Are there any reasons for using a custom defconfig instead of the upstream ones, or is this some historical thing related to the legacy vs mainline drivers?

The in-tree defconfigs e.g. `sources/meta-sunxi/recipes-kernel/linux/linux-mainline/arm/defconfig` contain a lot of additional random drivers e.g. for USB cameras, but are missing more basic drivers like audio:

```shell
root@bananapi-m2-berry:~# zcat /proc/config.gz | grep -E 'SND_SUN.I'
root@bananapi-m2-berry:~# 
```

... or hardware crypto:
```shell
root@bananapi-m2-berry:~# zcat /proc/config.gz | grep -E 'CRYPTO_DEV_SUN.I'
CONFIG_CRYPTO_DEV_SUN4I_SS=y
# CONFIG_CRYPTO_DEV_SUN4I_SS_PRNG is not set
# CONFIG_CRYPTO_DEV_SUN8I_CE is not set
# CONFIG_CRYPTO_DEV_SUN8I_SS is not set
```

In my opinion, it would be much better, especially from a maintenance perspective, to use the upstream defconfigs so that we can easily get up-to-date and more "complete" configs that match the kernel version.
This would, however, also mean that we use mainline DRM drivers by default.
We can still build additional modules using in-tree config fragments, if we want to.

Please let me know what you think. I could prepare a branch for testing.

nandra, 2025-03-24:
@okaestne I think that would be the right approach you describe above. Please prepare changes for PR. Thanks for your contributions.

okaestne, 2025-03-24:
@nandra I hope you are prepared for some breakage 😅 How important are the older kernel versions to you? Can we clean up to the latest v6.6?

nandra, 2025-03-24:
In master we should keep just what is necessary. Older kernel releases are working on older branches so it should be fine.

closed 2025-05-30 by nandra