linux-sunxi/meta-sunxi #446: Issue with WKS_FILE Not Found When Building Image for NanoPi NEO2

fabiocabaceira, 2025-04-03:
Hi everyone,

I'm building a Yocto image for the NanoPi NEO2 using the meta-sunxi layer, but I keep encountering an issue during the do_image_wic task. The error message I receive is: 

`  ERROR: custom-image-1.0-r0 do_image_wic: No kickstart files from WKS_FILES were found: custom-image.wks custom-image.wks. Please set WKS_FILE or WKS_FILES appropriately.`

I'm not sure what to do next. I need my image to generate a .wic output so I can flash it onto the NanoPi NEO2 using the SD card method. Any guidance on how to properly configure WIC for this board would be greatly appreciated!

nandra, 2025-04-03:
Hi @fabiocabaceira . Thanks for reaching out. Can you please share local.conf? wks file is set by this variable: https://github.com/linux-sunxi/meta-sunxi/blob/cabb64ab12be6be4704328f64457ef0ba710e2d4/conf/machine/include/sunxi.inc#L44C1-L44C41

Do you maybe override it in your layer? Because error says `No kickstart files from WKS_FILES were found: custom-image.wks custom-image.wks`. Thanks.

fabiocabaceira, 2025-04-04:
Hi @nandra 

Thanks for the response!

I'm using nanopi-neo2.conf, which includes sun50i.inc, and that in turn requires sunxi64.inc, but sunxi.inc is never included in this chain. Based on my understanding, the WKS_FILES variable from sunxi.inc wouldn't apply here.

Here is the relevant part of my local.conf : 
`IMAGE_FSTYPES += "ext3 tar.gz sunxi-sdimg wic.gz wic.bmap"`

I haven't explicitly set WKS_FILES in local.conf, and my image build currently fails with:
`ERROR: custom-image-1.0-r0 do_image_wic: No kickstart files from WKS_FILES were found: custom-image.wks custom-image.wks.`

Could it be that sunxi64.inc is missing a default WKS_FILES assignment? Or do I need to manually specify a .wks file for nanopi-neo2?

Appreciate any guidance you can provide!

Best,
Fábio Cabaceira

fabiocabaceira, 2025-04-04:
Hi @nandra ,

Thanks for your help. I was able to solve the issue and build my image. After the process was done, I ran the following command:

```
megafabio@FABIO-CABACEIRA-CDR:/home/megafabio/nanopi2/builder/build/tmp/deploy/images/nanopi-neo2$ sudo bmaptool copy mycustomeimage-image-nanopi-neo2.wic.gz /dev/sdb
[sudo] password for megafabio: 
bmaptool: info: discovered bmap file 'mycustomeimage-image-nanopi-neo2.wic.bmap'
bmaptool: info: block map format version 2.0
bmaptool: info: 868711 blocks of size 4096 (3.3 GiB), mapped 489535 blocks (1.9 GiB or 56.4%)
bmaptool: info: copying image 'mycustomeimage-image-nanopi-neo2.wic.gz' to block device '/dev/sdb' using bmap file 'mycustomeimage-image-nanopi-neo2.wic.bmap'
bmaptool: info: 100% copied
bmaptool: info: synchronizing '/dev/sdb'

```

**However, when I connect using the serial port, it reads nothing, and I can't find it on the network either. Do you have any suggestions on what might be causing this issue?**

Im reading serial using: 
`sudo minicom -D /dev/ttyUSB0 -b 115200`

[image]

nandra, 2025-04-04:
Unfortunately I don't have a board to test. But your commands + setup seems to be correct. It could be that u-boot is broken somehow if there is no serial output. On thing to check is if power supply is OK and also if on serial console RX/TX are not swapped. 

BTW: what was an issue with wks? Thanks.

fabiocabaceira, 2025-04-04:
@nandra The issue with the .wks file was indeed caused by an override in my layer.

I'm sharing my local.conf file below so you can review my configuration and check if there's anything incorrect.

```
# diskmon
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    HALT,${TMPDIR},100M,1K \
    HALT,${DL_DIR},100M,1K \
    HALT,${SSTATE_DIR},100M,1K \
    HALT,/tmp,10M,1K"

# extra-users
INHERIT += "extrausers"
PASSWD = "<REDACTED_PASSWORD>"
EXTRA_USERS_PARAMS = "usermod -p '${PASSWD}' root;"
INHERIT += "create-spdx"
SPDX_PRETTY = "1"

# nanopi-image
IMAGE_FSTYPES += "ext3 tar.gz sunxi-sdimg wic.gz wic.bmap"
WKS_FILE = "sunxi-sdcard-image.wks.in"
WKS_FILE_DIR = "${TOPDIR}/../layers/meta-sunxi/wic"
SUNXI_BOOT_SPACE ?= "40"
WKS_FILE_DEPENDS ?= "virtual/kernel u-boot"
IMAGE_BOOT_FILES ?= "${KERNEL_IMAGETYPE} boot.scr"
SPL_BINARY ?= "u-boot-sunxi-with-spl.bin"
IMAGE_CLASSES += "sdcard_image-sunxi"

# standard
CONF_VERSION = "2"
BB_NUMBER_THREADS = "${@oe.utils.cpu_count()}"
PACKAGE_CLASSES = "package_ipk"
USER_CLASSES ?= "buildstats"
BAD_RECOMMENDATIONS:append = " rng-tools"

# build-info
INHERIT += "image-buildinfo"
GIT_DIR = "--git-dir=${TOPDIR}/../.git"
GIT_BRANCH="${@oe.utils.getstatusoutput("git ${GIT_DIR} rev-parse --abbrev-ref HEAD")[1]}"
GIT_COMMIT="${@oe.utils.getstatusoutput("git ${GIT_DIR} log -1 --pretty=format:%h")[1]}"
GIT_DATE="${@oe.utils.getstatusoutput("git ${GIT_DIR} log -1 --date=short --pretty=format:%cd")[1]}"
GIT_VERSION="${GIT_COMMIT} | ${GIT_BRANCH} | ${GIT_DATE}"
GIT_BUILDER = "<REDACTED_GIT_INFO>"
BUILD_DATE = "${@oe.utils.getstatusoutput('date +%Y-%m-%d')[1]}"
IMAGE_BUILDINFO_VARS:append = " IMAGE_LINK_NAME GIT_BUILDER GIT_VERSION BUILD_DATE MACHINE"

# ffmpeg
LICENSE_FLAGS_ACCEPTED += "commercial"

MACHINE ??= "nanopi-neo2"
DISTRO ??= "poky-custom"
BBMULTICONFIG ?= ""

```

closed 2025-04-23 by nandra