linux-sunxi/meta-sunxi #448: Failure building Trusted Firmware-A (TF-A) in Yocto for Allwinner (sun50i_a64) with Link-Time Optimization

javoerro, 2025-04-13:
I’m trying to compile TF-A using Yocto on a Linux Mint machine, targeting the Allwinner (sun50i_a64) platform. However, the build fails with multiple “undefined reference” errors and “plugin needed to handle lto object” warnings when linking the bl31.elf binary. This error interrupts the overall image build for the Pine64.

#### Environment
Local OS: Linux Mint
Toolchain / Cross-compiler: aarch64-poky-linux-gcc and binutils provided by the Yocto toolchain
Yocto / meta-layer version: (Please replace with the specific version if relevant, e.g. “kirkstone,” “dunfell,” etc.)
Target: Pine64 (Allwinner A64), compiling “trusted-firmware-a” v2.12.1

#### Steps to Reproduce
Configure the Yocto environment with the appropriate layers for the Pine64 platform.
Set in `local.conf`:
```
MACHINE = "pine64-plus"
```
Start the build, for example:
```
bitbake trusted-firmware-a
```
Observe the process until it fails in the do_compile task of the “trusted-firmware-a” recipe.

#### Observed Behavior
The build process stops with linker errors:
- “undefined reference” to various symbols (e.g., handle_sysreg_trap, inject_undef64, plat_ic_get_pending_interrupt_type…)
- Repeated messages indicating “plugin needed to handle lto object”
- Ultimately, “ld: error: … undefined reference to…” and “ERROR: oe_runmake failed” cause the build to terminate.

```
...
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/arm_arch_svc_setup.o: plugin needed to handle lto object
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/std_svc_setup.o: plugin needed to handle lto object
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/std_svc_setup.o: plugin needed to handle lto object
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/bl31_entrypoint.o: in function `bl31_entrypoint':
| (.text.asm.bl31_entrypoint+0x110): undefined reference to `bl31_setup'
| aarch64-poky-linux-ld: (.text.asm.bl31_entrypoint+0x114): undefined reference to `bl31_main'
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/cpu_data.o: in function `_cpu_data_by_index':
| (.text.asm._cpu_data_by_index+0x8): undefined reference to `percpu_data'
| aarch64-poky-linux-ld: (.text.asm._cpu_data_by_index+0xc): undefined reference to `percpu_data'
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/context.o: in function `end':
| (.text.asm.el3_exit+0x30): undefined reference to `per_world_context'
| aarch64-poky-linux-ld: (.text.asm.el3_exit+0x34): undefined reference to `per_world_context'
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/runtime_exceptions.o: in function `sync_handler64':
| (.text.asm.sync_exception_handler+0x68): undefined reference to `rt_svc_descs_indices'
| aarch64-poky-linux-ld: (.text.asm.sync_exception_handler+0x6c): undefined reference to `rt_svc_descs_indices'
| aarch64-poky-linux-ld: /{WORKSPACE_DIR}/build/tmp/work/pine64_plus-poky-linux/trusted-firmware-a/2.12.1/build/sun50i_a64/release/bl31/runtime_exceptions.o: in function `sysreg_handler64':
...
```

#### Expected Behavior
The do_compile task completes successfully, producing the bl31.elf binary without linker errors, so the image build can continue.

#### Additional Information / Clues
This appears to be an **LTO (Link-Time Optimization) issue**. The compiler uses `-flto`, but the linker or binutils may not have (or may not be referencing) the necessary LTO plugin.

#### Questions / Help Needed
1.  Has anyone encountered a similar error compiling TF-A on Allwinner with LTO enabled?
2.  Is there a recommended Yocto recipe configuration to disable LTO or force LTO plugin detection in binutils?
3.  Is it preferable to use a different version of TF-A or a patch for the recipe that resolves the LTO conflict?

Any guidance on resolving these errors and successfully compiling TF-A within Yocto on Linux Mint would be greatly appreciated! Thank you in advance for your help.

nandra, 2025-04-14:
@javoerro which branch do you use for building please?

nandra, 2025-04-28:
@javoerro I was able to reproduce it locally also. I'm looking for solution.

nandra, 2025-05-30:
@javoerro I asked on meta-arm mailing list but no response so far. One possible workaround I found is following (you can set it in your local.conf):

```

closed 2025-06-10 by nandra